rust-embedded/embedded-hal #4: `Timer` API

japaric, 2017-06-09:
- What is missing in this API?
- Can this API be implemented for different devices?

hannobraun, 2017-10-27:
I've recently released [register mappings for LPC82x](https://crates.io/crates/lpc82x) and have now begun writing a HAL on top of that. I've started with implementing `Timer` for the LPC82x's WKT (Self-wake-up timer; chapter 18 in the [user manual](https://www.nxp.com/docs/en/user-guide/UM10800.pdf), if you're interested). I've had some problems/questions/ideas along the way.

### Modeling hardware capabilities

The WKT's purpose is to wake the chip up from the various sleep modes. It is very simple: You set a value, it counts down, it triggers an interrupt and/or sets a flag once finished.

Due to its simplicity, the hardware can't support the following methods:
- `get_timeout`: It doesn't remember the original value.
- `restart`: Since it doesn't remember the original value, I can't restart from it. I can only set a new one.
- `pause`/`resume`: If it's counting, it's counting. I can stop, but not resume from the same place.

It should be straight-forward enough to emulate these capabilities in software, and in my case I can't really see a reason why not to. But would this be the right thing to do in general, or should a HAL implementation represent the capabilities of the underlying hardware?

Software emulation should be fine, if the `Timer` implementation is expected to be the sole owner of its peripheral. But there might be use cases where it isn't, and then keeping state in software might cause trouble.

An alternative would be to split `Timer` into finer-grained traits. For this specific case this might not be worth the trouble, but there are other problems that could benefit from more focussed traits.

### Repeating timers

It seems to me that a lot of timer hardware can or should be used as a repeating timer. You set the value, it counts down. Once it has finished, the countdown starts again. For these cases, it would be useful to have a method that tells me whether the counter has reset since I last called that method.

I don't know if all repeating timer can support such a method. If not, it may need to be added to a separate trait.

### Stopping the timer

There isn't a method to stop the timer. There's `pause`, but that doesn't seem applicable. As mentioned above, `pause` needs software emulation on LPC82x's WKT, but `stop` (or `reset`) could be natively supported there.

### Starting the timer

Should `set_timeout` start the timer immediately? The lack of a `start` method seems to imply so, or is `restart` meant for that? This is especially relevant  for me, since the LPC82x's WKT automatically starts the timer after I write the timer value.

Again, the question is, should I emulate the separate steps for `set_timeout` and `start` in software, or is adding additional traits a better solution? A third solution would be to explicitely not specify this and leave it to the implementation, but that seems highly undesirable.

### Knowing whether the countdown is running

Do we need a method that tells us whether the timer is currently running? How would such a method relate to `wait`? Maybe `wait` could be extended to return the precise state (not started, running, finished), but I don't know how that would interact with `nb::Result`. I also don't know if timer peripherals can generally be expected to provide that information, so maybe `is_running` needs a separate trait.

### Interrupts

I have a `sleep` function built on top of my WKT-based `Timer` implementation. The only thing it needs to do that isn't already covered by `Timer` is enabling the WKT interrupt. I'm aware that configuration is considered to be out of scope for this crate, but I don't know if this case should be covered by that.

I might be sleeping in different ways for different reasons throughout my program, so enabling/disabling the interrupt could be considered part of the normal operation, not strictly initialization. Or I might want to write a library that provides a sleep function. It would be a shame if such a library had to depend on platform-specific crates.

Something like this could work:
``` rust
// in module `timer`

pub trait Interrupt {
    fn enable_interrupt(&mut self);
    fn disable_interrupt(&mut self);
}
```

### Measuring time

`Timer` seems specifically geared towards waiting for a specific amount of time to pass, but what about measuring time? Not all timers are suited for that (the LPC82x's WKT isn't), but some (many, most?) are. A `get_value` method would be required to do that.

If we're providing that functionality, it would make sense to also provide an additional convenience method that returns the passed time since the start of the timer.

### Conlusion

Most of the issues I mentioned could be solved in two ways:
- Fine grained traits that represent specific hardware capabilities.
- One trait, with software emulation for anything that hardware can't support.

I'm leaning towards fine-grained traits as a solution, but I'd like to hear other's opinions.

I'm willing to work on solving all issues I mentioned. Before I cook up a pull request, I'd like to come to an agreement over the general approach here.

japaric, 2018-01-15:
Thanks for your comment, @hannobraun. I haven't seen it until now.

It sounds like we should reduce the complexity of the `Timer` trait so it can properly fit your WKT
peripheral. I think this reduced version should work for both of us:

``` rust
trait Timer {
    type Time;

    /// Sets a timeout and starts the count down
    ///
    /// # Contract
    ///
    /// `self.set_timeout(timeout); block!(self.wait())` MUST block for AT LEAST the requested
    /// timeout
    fn set_timeout<T>(&mut self, timeout: T)
    where
        T: Into<Self::Time>;

    /// Waits until the timer times out
    ///
    /// # Contract
    ///
    /// - if `Self: Periodic`, the timer will start a new timeout right after the last one
    /// finished.
    /// - otherwise the behavior of calling `wait` after the last call returned `Ok` is UNSPECIFIED
    /// (implementers are suggested to panic on this scenario to signal a programmer error)
    fn wait(&mut self) -> nb::Result<(), !>;
}

trait Periodic {}
```

With this trait we can still implement e.g. the read with timeout abstraction which is one of the
use cases for the `Timer` trait:

``` rust
enum Error<E> {
    Timeout,
    Other(E),
}

fn blocking_read<S, T, N>(serial: &mut S, timer: &mut T, timeout: N) -> Result<u8, Error<S::Error>>
where
    S: serial::Read<u8>,
    T: Timer,
    N: Into<T::Time>,
{
    timer.set_timeout(timeout);

    loop {
        match serial.read() {
            Ok(byte) => return Ok(byte),
            nb::Error::WouldBlock => {} // check timer: see below
            nb::Error::Other(e) => return Err(Error::Other(e)),
        }

        if timer.wait().is_ok() {
            return Err(Error::Timeout);
        }
    }
}
```

> Stopping the timer

> Starting the timer

If there's an use case for this (i.e. if it lets you write generic code than you can't write with
just `Timer`) we can add separate traits for those two actions later on.

> Knowing whether the countdown is running

This is now specified in the contract. The timer is always active after `set_timeout` is called. If
it's not a periodic timer then it will enter the "expired" state after `timer.wait()` returns
`Ok(())`. Calling `wait` on an expired timer is unspecified but the recommendation is to panic! to
let the programmer know about the logic error.

> Interrupts

Can you elaborate on this `sleep` function? How is supposed to implemented on top of traits as a
generic function? My gut feeling is that the proposed `Interrupt` trait is too vague as it is as it
has no contract about what the interrupt should do, or when it is triggered.

> Measuring time

Measuring time using a timer that's not guaranteed to be a monotonic is error prone, IMO. If we want
to add something like this it should be a separate trait whose contract explicitly forbids also
exposing APIs like `set_timeout` which may reset / modify the counter.

Here's a draft of such a trait:

``` rust
/// # Contract
///
/// - `Self` MUST be a monotonically increasing timer whose counter has wraparound semantics (e.g.
/// DWT)
/// - `Self` MUST NOT expose an API to change its current counter value
trait MonoTimer {
    type Instant;
    // XXX this may need an associated `const WRAPAROUND_VALUE` because the counter may not wrap
    // around `Instant::MAX_VALUE` but at a lower value.

    /// Returns an instant corresponding to "now"
    fn now(&self) -> Self::Instant;
}
```

With an example use case:

``` rust
/// Times the execution time of the function `f`
fn timeit<M, T, F>(f: F, mono: &M) -> <M::Instant as Sub<M::Instant>>::Output
where
    F: FnOnce() -> T,
    M: MonoTimer,
    M::Instant: Sub<M::Instant>,
{
    let start = m.now();
    compiler_barrier();
    black_box(f());
    compiler_barrier();
    let end = m.now();
    end - start
}
```

hannobraun, 2018-01-16:
Hi @japaric,

thank you for the reply.

> It sounds like we should reduce the complexity of the Timer trait so it can properly fit your WKT
peripheral. I think this reduced version should work for both of us:

I like this suggested trait!

> > Stopping the timer
> >
> > Starting the timer
> 
> If there's an use case for this (i.e. if it lets you write generic code than you can't write with
just Timer) we can add separate traits for those two actions later on.

I don't have a use case currently. Automatically starting the timer on `set_timeout`, as per your suggested trait, works fine for me.

I agree that we can defer this until use cases pop up.

> > Knowing whether the countdown is running
> 
> This is now specified in the contract. The timer is always active after set_timeout is called. If
it's not a periodic timer then it will enter the "expired" state after timer.wait() returns
Ok(()). Calling wait on an expired timer is unspecified but the recommendation is to panic! to
let the programmer know about the logic error.

Sounds good to me!

> > Interrupts
>
> Can you elaborate on this sleep function? How is supposed to implemented on top of traits as a
generic function? My gut feeling is that the proposed Interrupt trait is too vague as it is as it
has no contract about what the interrupt should do, or when it is triggered.

To be honest, I can't quite remember what I was thinking back then. I agree with your gut feeling about the `Interrupt` trait. Maybe I'll remember later what I was going for. Until then, let's just assume it wasn't a good idea :-)

>  > Measuring time
>
> Measuring time using a timer that's not guaranteed to be a monotonic is error prone, IMO. If we want
to add something like this it should be a separate trait whose contract explicitly forbids also
exposing APIs like set_timeout which may reset / modify the counter.

I think forbidding the type from exposing such APIs is too strict. An API could easily prevent this interference by having an `Instant` type that contains a shared reference back to a `Timer`. The user of such an API could then call `Timer::now` and use the returned `Instant`s to measure time, and as long as an `Instant` exists, calling `set_timeout` or something like it would be impossible (assuming all functions that could interfere with measurements require `&mut self`).

Although I'm pretty sure you'd run into trouble when `Self::Instant` needs an explicit lifetime. This might require explicit support in `MonoTimer` (or whatever the trait will be called).

In any case, I don't think we need to come up with a solution right now. I don't have a use case for measuring time right now (I used to have one, but have since replaced the sensor that required this). I'd be happy to revisit this issue at a later point, once I have a use case again.

japaric, 2018-01-16:
> I like this suggested trait!

Let's go with it then.

> An API could easily prevent this interference by having an Instant type that contains a shared reference back to a Timer

That's a good alternative but implementation requires the [`generic_associated_types`](https://github.com/rust-lang/rfcs/blob/master/text/1598-generic_associated_types.md) feature, whose RFC has been accepted but has not been implemented yet. You want to write something like this, right?

``` rust
trait Monotonic {
    type Instant<'a>;
    fn now<'s>(&'s self) -> Self::Instant<'s>;
}

struct Timer { .. }
struct Instant<'t> { instant: u32, timer: &'t Timer }
impl Monotonic {
    type Instant<'t> = Instant<'t>;
    fn now(&self) -> Instant { /* .. */ }
}

let mut timer: Timer = /* .. */;
let start = timer.now(); // this freezes `timer`
// ..
let end = timer.now();
let elapsed: Ticks = end - start;
```

Anyhow, the different options for an Instant API can be discussed in another issue.

japaric, 2018-01-20:
The `timer::CountDown` is now part of a crates.io release so I'm going to close this. If you encounter any problem implementing / using that trait feel free to open a new issue!

There are other timer related traits that we'll probably want to add in the future; those should have their own RFC issues.

closed 2018-01-20 by japaric

hannobraun, 2018-01-22:
@japaric 
> You want to write something like this, right?

Yes, that's exactly what I was thinking. I plan to look into this more closely at a later date. Maybe `generic_associated_types` will be available by then.