rust-embedded/embedded-hal #5: `Capture` API

japaric, 2017-06-09:
- What is missing in this API?
- Can this API be implemented for different devices?

japaric, 2017-06-09:
> What is missing in this API?

Possibly methods to choose the capture "edge". Should this capture the rising edge or the falling edge? This is useful in generic context to capture on "high" signal: you want to capture the rising edge and then the falling edge.

Another question: is a `enum Edge { Rising, Falling }` enough, or should `Edge` be an associated type of the `Capture` trait? Are there microcontrollers that have a "capture both edges" setting?

adamgreig, 2017-06-14:
> Are there microcontrollers that have a "capture both edges" setting?
Yep, I think most STM32F3/4/7 and maybe 0 have options for rising/falling/both (along with a whole heap of other options...). See CCER register.

japaric, 2017-06-15:
OK. The F103 only has two options: rising and falling. The F303 has three: rising, falling and both. I would prefer if each device crate didn't end up inventing their own `Edge` enum so perhaps we can have the most common `Edge` enums in this crate. At least `Edge2 { Rising, Falling }` and `Edge3 { Rising, Falling, Both }`. Then we have to make the edge related API generic around the `Edge` enum. So maybe:

``` rust
trait CaptureOn<Edge> {
    fn capture_on(_: Edge);
}
```

then generic APIs can use a bound like `C: Capture + CaptureOn<Edge2>` when they need to capture the rising and falling edges in their implementation.

japaric, 2018-01-20:
Update: this trait is available in release v0.1.0 behind the "unproven" Cargo feature.

Fomys, 2022-07-31:
Hello,
This is my first time contributing here, I don't know if I'm doing it the right way.

With this API, I think it would be difficult to write drivers.
Some drivers only need a capture on the rising edge, which enum should they use? Edge2? Edge3?

Maybe a better way to handle this would be to use multiple traits :
```rust
trait CaptureRising {
    fn capture();
}
trait CaptureFalling {
    fn capture();
}
trait CaptureBoth: CaptureRising + CaptureFalling {
    fn capture();
}
```

And it will be used like this:
```rust
struct MyDevice<C: CaptureRising> {
    capture: C
}

impl<C: CaptureRising> MyDevice {
    fn something(&self) {
        self.capture.capture();
    }
}
```

With this, the driver will be compatible with all devices which implement `CaptureRising`.