rust-embedded/embedded-hal #8: `Qei` API

japaric, 2017-06-09:
- What is missing in this API?
- Can this API be implemented for different devices?

japaric, 2018-01-20:
Update: this trait is available in release v0.1.0 behind the "unproven" Cargo feature.

leshow, 2019-11-05:
I wrote a library for a simple rotary encoder library, and after posting about it in the rust-embedded/awesome-rust thread, someone directed me to this trait, that I had no idea existed.

I'm similarly having trouble for seeing how to map to the methods provided in the trait. Particularly the `direction` method.

`Direction` in `embedded-hal` has only 2 variants, and the `direction()` function returns just `Direction`. However, in my case, it's possible there is no change in direction, and the `Count` type itself has no inherent direction. The only thing I can think of is returning Updirection if the count is above zero and Down in the other case, but that doesn't really provide anything useful for a rotary encoder

It's very possible I'm just not understanding something, but I can't see how it fits together at all.