rust-embedded/embedded-hal #15: "Mutex" Trait based API

vitiral, 2017-06-16:
This is a counter proposal to #14 and aims to be simpler and more future proof.

I have sketched out an implementation which compiles (but is not tested) here:

https://github.com/vitiral/utex/blob/master/src/lib.rs

## Reason of need
The embedded ecosystem is very likely to have *multiple* ways of allocating data, from statically to managed-static to full dynamic. However, most libraries should not care who allocated data -- only that it will not get freed and is not being used by others.

Therefore there is a strong need for a single trait that all of these libraries can use for wrapping the data they return.

## Issue at Hand
The existing `Buffer` API will tie us to the following:
- a specific buffer type and way of representing data
- a lock/unlocking mechanism that can not be easily extended by other crates. For instance, it would be difficult to implement a memory manager using the Buffer API

This design aims to address this by providing a `Mutex` *trait* which can then be implemented in any way we please. Something like the `Buffer` proposed would be free to implement this trait and then be used for the full API.

japaric, 2018-02-14:
I'm closing this since #13 was also closed

closed 2018-02-14 by japaric

vitiral, 2018-02-15:
sorry, I should have closed this before. Thanks!