rust-embedded/embedded-hal #25: Release on crates.io

hannobraun, 2017-11-29:
What's currently blocking the release of this crate on [crates.io](https://crates.io/)? One obvious blocker is that [nb](https://github.com/japaric/nb) is not yet published there. I've [opened an issue](https://github.com/japaric/nb/issues/9) there too.

japaric, 2017-12-01:
> What's currently blocking the release of this crate on crates.io?

We are missing digital I/O traits (I smuggled in an OutputPin trait while no one was looking, though) and those are like the most common kind of I/O.

I'd like to see a resolution to RFC #18 so we can define the module hierarchy in this crate. (Or perhaps the blocking / higher level traits should be in a different crate, IDK)

We need to figure out some way to indicate the stability of the traits (IDK if `#[unstable]` is now available for user libraries) so we can distinguish between "this is our first stab at an abstraction for this peripheral, anything goes" and "we are 50%+ sure this abstraction won't suffer a breaking change the next week". We could use a non-default "unstable" Cargo feature; put all the unstable API behind that feature and reserve the right to break that unstable API in patch releases.

hannobraun, 2017-12-04:
Thanks for the info!

> We need to figure out some way to indicate the stability of the traits (IDK if `#[unstable]` is now available for user libraries)

It doesn't seem to be. There's an [open issue](https://github.com/rust-lang/rfcs/issues/1491).

> We could use a non-default "unstable" Cargo feature; put all the unstable API behind that feature and reserve the right to break that unstable API in patch releases.

I think that sounds like a workable solution.

closed 2018-01-16 by japaric