rust-embedded/embedded-hal #40: `blocking::serial::*` API

japaric, 2018-02-15:
These traits became available in release v0.1.2 behind the "unproven" Cargo feature.

hannobraun, 2018-02-16:
@japaric Did you add the `needs-impl` tag on purpose? It doesn't seem to apply, since the trait has a default implementation. Or do you want to see alternative implementations that bypass the default one before marking the API as proven?

japaric, 2018-02-16:
> It doesn't seem to apply, since the trait has a default implementation.

The default implementation is opt in. I still think it should be checked that the default implementation works on hardware.

> Or do you want to see alternative implementations that bypass the default one before marking the API as proven?

That would be ideal but I don't think it should be a hard requirement.

hannobraun, 2018-02-19:
> The default implementation is opt in. I still think it should be checked that the default implementation works on hardware.

Ah, that makes sense.

I'm using the default implementation in [LPC82x HAL](https://github.com/braun-robotics/rust-lpc82x-hal/blob/c475f38d3322bb2adeabd572ff2408ee27ac7d97/src/usart.rs#L285). As far as I can tell, everything works.