rust-embedded/embedded-hal #41: `digital::InputPin` API

japaric, 2018-02-15:
This trait became available in release v0.1.1 behind the "unproven" Cargo feature.

therealprof, 2018-02-25:
Implemented in nrf51-hal = "0.4.1" and used in microbit = "0.4.1".

Nemo157, 2018-03-06:
Would this be the correct trait to include some sort of simple edge detection API (e.g. for buttons hooked up to GPIO)?

Caemor, 2018-09-17:
Does this api still needs more drivers?

It seems to be quite stable since it's introduction half a year ago.

Caemor, 2018-09-18:
The only problem I found with this api since using it is that errors can't get propagated any further and need to be converted to a default value. 

E.g. the linux sys_fs can receive an file error while trying to read the value (https://github.com/rust-embedded/rust-sysfs-gpio/blob/master/src/lib.rs#L337) and this can't be handled by a simple bool.

eldruin, 2018-09-21:
I also stumbled at the same problem as @Caemor with the `OutputPin` trait as well. In an I2C I/O expander driver I am writing I want to provide access to the individual pins as a bunch of `Px` structs implementing `OutputPin` and `InputPin`. At the moment I saw no other choice but to [panic if there was any communication problem](https://github.com/eldruin/pcf857x-rs/blob/a1ea28f9b333d066fe44f1453e00f3fca17c3121/src/pins.rs#L81). I was about to implement the `InputPin` traits and saw that I would have the same problem again.

Maybe we should discuss this in a separate issue?

ryankurte, 2018-09-23:
Interesting problems! I think as this spans all operations (not just `Digital::InputPin`) another issue would be great ^_^

nils-van-zuijlen, 2019-11-14:
If this API does not need driver nor implementation, when will it be proven ?

therealprof, 2019-11-14:
The problem is: There's no process to mark something as proven. There's a new RFC open #163 in attempt to completely revamp the way we get things added/changed.

Since there're enough implementations already we could simply accept a PR to mark it proven.

rubberduck203, 2020-02-02:
I think this issue can be closed.

therealprof, 2020-02-02:
Fair enough.

closed 2020-02-02 by therealprof