rust-embedded/embedded-hal #59: Shared type definitions (e.g. MHz, bps...)

nordmoen, 2018-03-08:
I want to create this issue to discuss a shared place to store necessary hardware type definitions. These could be stored in the `embedded-hal` crate, but could also easily be in their own crate. I just think we should discuss a shared place to store these outside the processor specific crates.

There are many type definitions that can be useful when working with low-level sensors. At the moment there are no shared place that such type can or are stored. A couple of very handy types are currently in [`stm32f30x_hal::time`](https://docs.rs/stm32f30x-hal/0.1.2/stm32f30x_hal/time/index.html) (and also replicated in [`stm32f103xx_hal`](https://japaric.github.io/stm32f103xx-hal/stm32f103xx_hal/time/index.html)) for dealing with time.

An easy first step is to create a `type` module for `embedded-hal` and extract the types defined in `stm32f30x_hal::time` and potentially add more later if necessary.

I'm not sure if there are other definitions apart from time that should be abstracted out. I don't think this issue should be about physical types (e.g. `cm` or `m/s`), but should focus on shared types that are necessary for interacting with hardware.

thejpster, 2018-03-08:
For most physical types there are other crates, so as the
docs.rs/measurements crate that I manage.

I think frequency, bitrate and time should be in embedded-hal though as
most APIs will need them.

nordmoen, 2018-03-08:
Not to derail the discussion after the first comment, but would it be possible to transition some of the time handling to use [`core::time::Duration`](https://doc.rust-lang.org/nightly/core/time/struct.Duration.html) instead of raw `u32`s? I understand that the additional `u64` is not wanted, but for subsecond durations shouldn't it be optimized away?

The advantage would be more interoperability with the rest of the ecosystem, and also many helpful functions to ensure no overflow etc.

therealprof, 2018-03-08:
@thejpster Great stuff! I was actually looking for that, partially also for #32.
@nordmoen I fully agree that the base units required by `embedded-hal` should be part of it instead of being duplicated in every single implementation. I'm a bit torn about `core::time::Duration` because I'm not exactly sure how much baggage that would add; guess we'll have to just have to try and see.

hannobraun, 2018-03-08:
Partially related discussion: #24 
See also (from the linked issue): https://github.com/TheZoq2/embedded-hal-time

Nemo157, 2018-03-08:
> Not to derail the discussion after the first comment, but would it be possible to transition some of the time handling to use `core::time::Duration` instead of raw `u32`s? I understand that the additional `u64` is not wanted, but for subsecond durations shouldn't it be optimized away?

It would be very likely that it won't be optimized away unless the `CountDown` implementation (or whatever is using the `Duration`) gets fully inlined to where the `Duration` is created. (This does appear to happen in my current prototype that's just blinking an LED based on a `CountDown`, but I don't know that we could rely on it in larger projects).

I was just thinking about whether a `ShortDuration { us: u32 }` or something similar should be available in `embedded-hal` for implementations to use. I would assume most usage of `CountDown` would be fine with that as the range.

Kixunil, 2018-03-13:
There are cases when long times don't make sense. I'd be against calling it `ShortDuration`, but prefer defining it as `Microseconds(pub u32)` instead.

jonas-schievink, 2018-03-27:
To name a few alternatives to the `measurements` crate (which forces the use of `f64`, which might be suboptimal for small µCs), there's [dimensioned](https://github.com/paholg/dimensioned) and [uom](https://github.com/iliekturtles/uom), both of which support `#![no_std]` and allow using arbitrary types.

therealprof, 2018-03-27:
Yeah, it's a somewhat odd choice to use f64 for a number of reasons, but especially without FPU it does hurt. I was actually going to look into what was required to change that, potentially using https://github.com/japaric/fpa .

kellerkindt, 2018-08-07:
I am trying to implement a new device driver that should ideally only depend on ```embedded-hal``` and I found this issue while searching for a solution to the problem below:

For it to work, I need to wait up to a certain amount of microseconds for an input pin to change, check whether it has changed and wehther it is now in the desired state. This can be implemented for example using the [MonoTimer of stm32f103xx-hal](https://github.com/japaric/stm32f103xx-hal/blob/master/src/time.rs#L80). Once the current 'time' / [Instant](https://github.com/japaric/stm32f103xx-hal/blob/master/src/time.rs#L108) is read a simple while loop could check whether the input changed, as long as time timeout has not reached yet (```.elapsed() < us * (.frequency() / 1_000_000)```).
Since a blocking behaviour might not be desired - as well as the dependency on [stm32f103xx-hal](https://github.com/japaric/stm32f103xx-hal/blob/master/src/time.rs#L80) - one might think the [CountDown](https://github.com/rust-embedded/embedded-hal/blob/master/src/timer.rs#L60) could be used. But it requires the Time type to be known - which for the stm32f103 is defined in the [stm32f103xx-hal as Hertz](https://github.com/japaric/stm32f103xx-hal/blob/a25fb6276a81b51528761fa44d42df89cf6034e0/src/timer.rs#L49).
Both strategies require a dependency on the device-hal crate which I would like to omit. Am I missing and easy / obvious approach?

fionawhim, 2019-07-25:
Hey, I wanted to revive this stalled issue. I’m writing a device driver that would really like a `CountDown` in milliseconds, so I’m looking for a common `Time` type that can represent that.

Has anything changed re: `core::time::Duration` in the past year that would make it more or less attractive to adopt in `embedded-hal`?

little-arhat, 2019-07-27:
There is also https://crates.io/crates/bitrate

wbuck, 2021-01-06:
@kellerkindt or @fionawhim did either of you ever come up with a solution? 

I'm also writing a driver and found that I have to depend on the `stm32f3xx_hal`  device crate for the same reason you both mentioned.

wbuck, 2021-01-12:
@eldruin OK, I submitted a PR for integrating [embedded-time](https://github.com/FluenTech/embedded-time/) for the [stm32f3xx-hal](https://github.com/stm32-rs/stm32f3xx-hal) device crate.

It was a pretty smooth integration for the most part except for the `CountDown`/`Periodic` timers due to the limitations on their generic constraint `Into<Self::Time>`.

I want to see if there are some other devices crates that would be willing to accept a PR for this.

MathiasKoch, 2021-01-14:
I would love to see the same happen for https://github.com/stm32-rs/stm32l4xx-hal