rust-embedded/embedded-hal #66: stm32f7x-hal and stm32f7x crate

musitdev, 2018-03-26:
For a card I work on I started several crate for the stm32f7x processor:  [`stm32f7x-hal`](https://github.com/musitdev/stm32f7x-hal),  [`stm32f7x`](https://github.com/musitdev/stm32f7x) and an example [`rust-f7-example'](https://github.com/musitdev/rust-f7-example). There 're based on the other crate from @japaric. I use the STM32F7x.svd file to generate the stm32f7x crate. I didn't already registered them to crates.io because they are in a early stage. If you need it I'll do it.
I try to be compatible with the embedded-hal. GPIO is working, I work on the Serial / USART part. I test on a Nucleo-144 (STM32F746) board (the example crate).
I've made some change in the svd file because not all of its content was generated. The derivedFrom tab wasn't use for every declaration. There are still some errors during the generation. 
I have some reflexion about the use of svd file, mostly linked with the size of it and the code generated.

musitdev, 2018-03-26:
Thank for your help. I see the other crate but I want to use RTFM so I follow the SVD way. I have some problem with USART init. it's a good opportunity to learn to use gdb and learn more about STM32 processor.

musitdev, 2018-03-27:
I'm open. In my opinion, the svd is two big if you don't use all the processor features. A conditional generation or compilation of the generated code, can allow to share the same svd and use different features. Depending on the needs, the svd can be extended or features activated. The other solution ,simpler ,is to name the crate with the final card. There is card's specific code in the hal too (ex: port declaration) so the same problem will arise when using the hal. In this case on single crate for the card with hal, svd lib and example can be a solution.

therealprof, 2019-01-12:
@musitdev While checking for open tickets I stumbled across this one and noticed that it actually does not belong here. I hope you don't mind that I'm going to close it.

In case you're still working on this, the PACs provided by [`stm32-rs`](https://github.com/stm32-rs/stm32-rs) has turned out to be a great help for development of universal HAL impls and coincidently (or not) there's also a organisation with the same name where we bundle our STM32 efforts: https://github.com/stm32-rs. You're more than welcome to check it our and participate.

closed 2019-01-12 by therealprof