rust-embedded/embedded-hal #78: I2C driver guidance/best practices needed

wez, 2018-04-21:
An I2C master instance can be used to address multiple devices, but all of the drivers effectively take ownership of the I2C object and prevent sharing it with other drivers to talk to other devices connected on the same bus.

LSM303DLHC , MCP3425 , SGP30 and HTS221 which are all listed first on the https://github.com/rust-embedded/awesome-embedded-rust list move in the I2C instance.

https://docs.rs/si5351/0.1.5/si5351/ takes a mutable reference.

I think we need some guidance/best practice recommendations to allow addressing multiple slave devices.

Does it make sense to advise hal implementors to build in some kind of interior mutability for this?
Do we need some other token/type to manage ownership of the I2C bus for the duration of some IO?

wez, 2018-04-21:
A simple but sort of gross solution might be to keep passing in the I2C reference to each method of drivers that might need to do IO.

therealprof, 2018-04-21:
Hey @wez, I think #35 sums up the problem pretty well...

wez, 2018-04-21:
ah, I was looking for I2C in titles and missed that one... reading through it now :)

wez, 2018-04-21:
Closing in favor of the other issue

closed 2018-04-21 by wez