rust-embedded/embedded-hal #88: i2c: address format

chrysn, 2018-06-22:
I've encountered i2c addresses both in on-the-wire format (typically given as an even number, between 0 and 254, to which 1 is ORed for reads)  and in 7-bit format (a number between 0 and 127).

Which one of those is addr in the read/write/write_read functions supposed to be?

If it's 7-bit, I'd add a "Slave addresses are always given as 7-bit numbers (0 <= address < 128)." in the module documentation, and maybe change "`SAD+R` = slave address with 8th bit set to 1" to "`SAD+R` = slave address followed by bit 1" and similar at the function level. (Can provide a PR if that's the intention).

austinbes, 2018-06-22:
:+1: to making this point clear in docs

therealprof, 2018-06-23:
@chrysn Please be aware that potentially devices may use the long address format, too. I haven't seen any and we don't support addressing them at the moment but if we're clarifying the API that might be worth a mention. Please create a PR.

chrysn, 2018-06-23:
@therealprof, devices that use the other format need to convert the address either way. When using 7-bit style, implementations with 7-bit addresses will just use them, and others will do `addr << 1` for writes and `(addr << 1) | 0x01` reads. (If we went for 8-bit addresses, some could just use them and others would have to right-shift).

Or are you talking about 10-bit addresses? They are a completely different beast, but yes, I can add a note about them (haven't used them yet).

therealprof, 2018-06-23:
Talking about 10bit addresses here. I think there's no ambiguity about 7bit addresses except for the fact that the ones we take are the unshifted ones rather than how they appear on the bus.

closed 2018-06-26 by therealprof