rust-embedded/embedded-hal #90: Approve proven traits

therealprof, 2018-07-23:
I would suspect that there're a lot of candidates where the requirements for being a proven trait are now fulfilled. It would be great if we could collect a list of implementations to remove the `unproven` gate.

hannobraun, 2018-07-23:
Theoretically, there should be an open issue for each unproven trait, with tags to describe what's missing. See the [unproven-api](https://github.com/japaric/embedded-hal/issues?q=is%3Aissue+is%3Aopen+label%3Aunproven-api) tag.

I'm planning to go through all those issues and update the tags, but given my time budget for embedded-hal, that might take a while :)

It would also be great if someone could make sure that each unproven trait is covered by an open issue.

hannobraun, 2018-11-12:
@caemor
> I went through the repository and looked for all unproven traits. All points without a checkmark don't have their own unproven issue yet.

Thank you!

> * [x]  blocking::serial::* #40 - Is this already proven? As the unproven status only exists in the changelog and the issue itself but not in the code!

No idea.

> * [x]   Inputpin: #41 - Is waiting on #92, needs-impl and needs-driver labels could maybe be removed

I removed the tags.

> Missing implementations:
> 
> * [x]   Qei: #8 - Implemented + demo example in [stm32f103xx-hal](https://github.com/japaric/stm32f103xx-hal), no 2nd independent implementation was found in a quick search over the awesome embbeded rust list
> 
> * [x]   Pwm: #7 - Implemented + demo example in [stm32f103xx-hal](https://github.com/japaric/stm32f103xx-hal), no 2nd independent implementation was found in a quick search over the awesome embbeded rust list
> 
> * [x]   Capture: #5 - I couldn't find any impls and drivers

Did you find drivers using Qei and Pwn?

> Unproven Traits without issue:
> 
> * [ ]   Toggleable Outputpin
> 
> * [ ]   Stateful outputpin
> 
> * [ ]   Watchdog (#75, #98)
> 
> * [ ]   Adc (old: #10)
> 
> * [ ]   blocking::rng (maybe #38 could be used but it is missing the labels?)
> 
> * [ ]   blocking::spi::write_iter
> 
> * [ ]   blocking::i2c::write_iter
> 
> * [ ]   blocking::i2c::write_iter_read
> 
> 
> Should we open new issues for these? And maybe close a few of the old issues referenced in the last part?

Yes, opening issues for those is probably a good idea. If you want to do that, that would be very appreciated! If you need someone with permissions to close issues or add tags, feel free to ping me and tell me what to do. I don't have much time for deep-thinking work right now, but I should be able to find a minute here or there to follow simple instructions :)

eldruin, 2018-11-12:
I would like to write a driver using the PWM traits for PCA9585 and/or TLC5947 some time soon.

eldruin, 2020-01-06:
Follow up: I did write a driver for the [PCA9685 driver](https://github.com/eldruin/pwm-pca9685-rs) I could not implement either of the PWM traits because all the methods are infallible. I did not feel like starting another `digital::v2` debacle so I've just dropped it. Some more comments at #7

ryankurte, 2020-06-19:
As of the recent hal-team discussion and `1.0.0-alpha`, `unproven` has been removed and all traits are [fallible](https://github.com/rust-embedded/embedded-hal/blob/036cfa48e42b8c7ec78dcb919b8b0d24cf6939b6/src/pwm.rs#L111), which I believe resolves this issue 🎉 (Please feel free to re-open if I have overlooked something and this is not the case)

closed 2020-06-19 by ryankurte