rust-embedded/embedded-hal #98: Consider improving watchdog API using move semantics

hannobraun, 2018-09-30:
@astro [suggested the following](https://github.com/rust-embedded/embedded-hal/pull/76#issuecomment-417413406):

```rust
/// Feeds an existing watchdog to ensure the processor isn't reset. Sometimes
/// commonly referred to as "kicking" or "refreshing".
#[cfg(feature = "unproven")]
pub trait Watchdog {
    /// Triggers the watchdog. This must be done once the watchdog is started
    /// to prevent the processor being reset.
    fn feed(&mut self);
}

/// Enables A watchdog timer to reset the processor if software is frozen or 
/// stalled.
pub trait WatchdogEnable {
    /// Unit of time used by the watchdog
    type Time;
    /// The started watchdog that should be `feed()`
    type Target: Watchdog;
    /// Starts the watchdog with a given period, typically once this is done 
    /// the watchdog needs to be kicked periodically or the processor is reset. 
    ///
    /// This consumes the value and returns the `Watchdog` trait that you must
    /// `feed()`.
    fn start<T>(self, period: T) -> Self::Target where T: Into<Self::Time>;
}

/// Disables a running watchdog timer so the processor won't be reset.
#[cfg(feature = "unproven")]
pub trait WatchdogDisable {
    type Target: WatchdogEnable;
    /// Disables the watchdog
    ///
    /// This stops the watchdog and returns the `WatchdogEnable` trait so that
    /// it can be started again.
    fn disable(self) -> Self::Target;
}
```

I think this is an interesting approach that's worth thinking about.

luojia65, 2020-02-27:
This would work well on my HAL crate: https://github.com/gd32v-rust/gd32vf103-hal/blob/68a9d3b594038480504e3b52abea1a48aefaf148/src/wdog.rs
There would be two conditions:
1. The Watchdog after enabled is turned into another struct, like Watchdog<Enabled> or something. This could take ownership of old struct to prevent double `start` calls in compile time, and prevent `disable` calls if the watchdog is already disabled.
2. If we don't need to implement different traits and functions for a disabled watchdog and an enabled one, we may use `type Target = Self`, which can also be adapted to the older WatchdogEnable trait.

eldruin, 2020-07-15:
PR #222 by @luojia65 was merged and will be released in v1.0.0.
Can this be closed now?

hannobraun, 2020-07-16:
Yes, I think so.

closed 2020-07-16 by hannobraun