rust-embedded/embedded-hal #99: The future of `Unproven` traits and practices for new and breaking changes

ryankurte, 2018-10-02:
This issues is to discuss our management of breaking and non-breaking changes in the 0.x series of releases, to limi the scope of the discussion I suggest this is revised before we stabilise.

As mentioned in #97, the `unproven` flag on traits aren't working quite as we might intend.
We've also put forward an feature-flag approach to mitigate the impact of breaking changes to hal traits on library users that I propose we adopt for all future changes.

Please refer to #97 for an introduction to this approach and previous discussion about the `unproven` flag.

cc. @hannobraun @therealprof @eldruin

therealprof, 2018-10-03:
@ryankurte Maybe you should should lay out what the vision is regarding the feature-flag approach and create an RFC for the changes this requires.

Regarding the breaking and non-breaking changes and `unproven` flag: IMHO we should encourage people to create pull requests removing the `unproven` feature gate by providing references to known implementations which provide and/or use the `unproven` trait in question. We'll verify those claims, accept the PR, bump the version and release a new version. For breaking changes we should announce the upcoming change to solicit for more breaking changes and then do the same.

However I'm not quite convinced we really need to cause breakage for #97. I'd rather have separate traits for the fallible versions. That way people can easily adjust and bridge between the two versions instead of breaking everything in one swoop.

hannobraun, 2018-10-03:
> Yes, as a transition strategy.

Okay, understood. I'm open to discussing the transition strategy, especially since I recently realized we need to add those additional feature flags to the Travis build, then remember to remove them again later. Overall, I don't feel strongly about the transition strategy, as long as the end result is good.

closed 2019-04-12 by ryankurte