rust-embedded/embedded-hal #100: [RFC] Managing breaking changes to existing traits

ryankurte, 2018-10-04:
As highlighted in #95, #97 and #99, there are likely to be situations in which we need to fix or upgrade traits in breaking ways, and these are likely to have significant ramifications on the embedded ecosystem. This RFC proposes an approach to manage these breaking changes.

### Goals:
- Mitigate the downstream breakage caused by embedded-hal breaking changes
- Communicate breaking changes to give library uses an opportunity to update

### Approach:
We propose a staged approach for managing breaking changes, with some time between each stage to allow API consumers to update.

In the first release, replacement traits are introduced and feature-gated allowing users to opt in to their use. Feature names start with with `use-` and a description of the change, for example `use-fallible-digital-traits` for the new fallible digital trait update in #97. Existing traits are marked `#[deprecated]` to alert users to the need to update, and this opt-in flag should be added to the CI build matrix. Documentation should also be updated to recommend the use of the opt-in traits and the related github issue for the change. 
This stage alerts HAL users of deprecation and allows immediate migration should they desire, without requiring any explicit action.

In the second release, the opt-in feature flag is replaced with an opt-out flag to allow regression to the previous trait if required, this should start with `regress-`. Continuing the fallible digital trait example this would be `regress-infallible-digital-traits`. This opt-out flag should be added to the build matrix, and the previous flag removed. Documentation should be updated to mention opt-out trait availability if required.
This stage will break HAL consumers using the previous traits, requiring explicit action to update or defer the change.

In the third release, the old traits are removed, feature flags are removed from the package and the CI build matrix, and the documentation should be updated to remove mention of the opt-out trait. This stage will break any hal consumers that have not updated to the new traits.

In each release, these changes will be documented in the changelog (as is standard) and should be published through other useful means. The linking of the related issue in the documentation should allow for feedback from HAL consumers at each stage.

I also wrote a small tool to analyse the dependency requirements, resolutions, and features of published crates (https://github.com/ryankurte/rust-dep-check), which would let us track migration through versions and feature flags. An example output showing the current use of `embedded-hal`:

```
Found 100 crates using 'embedded-hal'
Version requirements:
	^0.1	(0.1.3):	   1 / 100 (1.00 %)
	^0.1.0	(0.1.3):	   6 / 100 (6.00 %)
	^0.1.1	(0.1.3):	   1 / 100 (1.00 %)
	^0.1.2	(0.1.3):	   7 / 100 (7.00 %)
	^0.2	(0.2.1):	  19 / 100 (19.00 %)
	^0.2.0	(0.2.1):	  12 / 100 (12.00 %)
	^0.2.1	(0.2.1):	  45 / 100 (45.00 %)
	~0.2	(0.2.1):	   9 / 100 (9.00 %)
Features:
	unproven:	  35 / 100 (35.00 %)
Resolved versions and features:
	0.1.3:	  15 / 100 (15.00 %)
		unproven:	   3 / 15 (20.00 %)
	0.2.1:	  85 / 100 (85.00 %)
		unproven:	  32 / 85 (37.65 %)
```

### Questions
- Should this apply to traits marked `unproven`? My view is that given the propensity of `unproven` traits we may as well try to mitigate all breakages or no worry about it at all (see the alternative).

### Alternative
Do nothing, semver already mitigates breaking changes and we’re in the 0.x series with no API stability guarantees. This is much simpler, though not particularly supportive of our API consumers or the possible impact of breaking changes on the ecosystem, and it doesn't give our consumers any notice of the change outside of the github discussions.

cc. @hannobraun @therealprof

jamwaffles, 2018-10-04:
I like the safety and caution of this approach, but it feel like it's more suited to post-1.0 quality software. What would the version numbers look like in the above process, semver-minor or semver-patch? I noticed you didn't define any timeframes for these releases. To keep things flowing, I think the deprecation process should be pretty quick at the stage the ecosystem is at now.

Personally, I'd prefer to have this stuff break catastrphically (as per the *Alternative* section) and fix it once, but I can see that causing a lot of pain for people with more embedded Rust written than myself.

ryankurte, 2018-10-04:
> [I] feel like it's more suited to post-1.0 quality software

I agree in principle, but I think if we want to get it right by 1.x we should start practicing it now and work out what works and doesn't. In addition, I suspect `embedded-hal` will not reach 1.0 for some time yet, and as the ecosystem grows this is going to become ever more important.

> What would the version numbers look like in the above process

Roughly following the semver spec for breaking changes, so for 0.x: patch, patch, minor, and for >1.x: minor, minor, major. The first stage is not breaking and should be updated automatically to notify consumers, the second only requires a feature flag change to opt out, the third explicitly breaks anything not updated.

I can see an argument that the second two stages should both be considered breaking, the flags will have to be updated to work, and I'm not completely convinced either way. It's also important to note that there is likely to be than one change applied per release.

> I noticed you didn't define any timeframes for these releases. To keep things flowing, I think the deprecation process should be pretty quick at the stage the ecosystem is at now.

Agreed that it should be quick. I'd like to see us iterating and releasing more regularly, and I think having some process like this would decrease the risk / scariness of releasing so we can do it faster. I think if we had a minimum limit of a couple of weeks between each stage we could have a similar release cadence.

This approach also lets us collect feedback on and learn from our releases / changes and the impact they have on our consumers, and to stop part way if there are issues, where breaking changes are otherwise not communicated except for the breakage.

hannobraun, 2018-10-05:
I'm generally in favor. The one concern I have is what consequences this has on CI. If I understand correctly, all combinations of features will have to be tested. Could it cause problems to have that many CI builds (i.e. running up against usage limits)?

> Roughly following the semver spec for breaking changes, so for 0.x: patch, patch, minor, and for >1.x: minor, minor, major.

I disagree with this part. Upgrading to a new patch (0.x) or minor version (>= 1.x) should never break the build (except to fix security issues). Even if the fix to the breaking change is as simple as adding a feature.

hannobraun, 2018-10-08:
> Fair enough, do you have a preferred arrangement, or would you be happy with 0.x: patch, minor, minor, and >=1.x: minor, major, major? While it deviates from strict semver for 0.x, it benefits us if the first deprecation marking is an automatic upgrade.

I'd be very happy with that. I don't see how it deviates from semver. Adding a deprecation won't break the build, unless the crate denies warnings, which is effectively opting out of semver stability.

eldruin, 2018-10-09:
This sounds very good @ryankurte.

1. It seems to me like there will be an increase of feature flags and it can become difficult to track the flags across releases.
In this regard, I would add a list somewhere where the flags are registered, including in which release they will appear/disappear. A more condensed form of the Changelog, if you like, to ease the release process.

2. We can have a first try at testing combinations of flags in the CI build in #97. I only did this locally.

hannobraun, 2018-10-11:
I just noticed @therealprof and @ryankurte having a discussion on IRC that they really should be having here, in my opinion :)

@therealprof said this:
> That still doesn't allow mixing and matching which is a huge problem IMHO.
> It requires the HAL implementation and all drivers to use the same version of the trait.

This brings up an issue that is hugely important in my opinion: You can't have multiple drivers among your dependencies that depend on different versions of the trait. If one driver enables the features, than all drivers that don't enable the feature are suddenly broken and won't build. Unless I miss something, there's no other way to fix the broken build than fork one group of drivers and change them to do what the other group of drivers is doing.

I think this makes all other advantages that this proposal has moot. What's the point of having a smooth transition, if everything breaks down the moment I use two drivers that are at different stages of the smooth transition?

In light of this, I now prefer the following approach:
- Add new trait variations under a new name (e.g. `MyTrait2`) deprecate the old trait (e.g. `MyTrait`) and publish a non-breaking release.
- On the next breaking release, replace `MyTrait` with `MyTrait2`, deprecate `MyTrait2`.
- On the next breaking release after that, remove `MyTrait2`.

Yes, that's one additional breakage, but at least it won't break my build if I mix two drivers.

That whole thought about mixing and matching drivers, has led me to another insight: It's possible to import different versions of the same package under different names: https://github.com/rust-lang/cargo/pull/4953

That means, a HAL implementation can release a transition version that supports both and old and a new version, allowing application authors to use any driver during the transition period.

therealprof, 2018-10-11:
After thinking about it some more and some discussion on IRC I have to revise my previous statement.

I see two fundamental problems with a part of this approach (the `replacement traits` and the feature gating of them):

1. In one application you can only have exactly one version of a trait and all components must adhere to that, i.e. the application, the HAL impl and all used drivers need to implement the exact same `embedded-hal` version with the same "version" of the trait. This can be annoying already when only the application and the HAL impl are involved but add a few drivers to the mix and this can easily turn into a blocker
2. Feature gating itself can be very annoying and should be used sparsely rather than generously. This begins with the question where to define the features (because removing a feature is a breaking change) and how to select them (if you have a hierarchy of crates there're multiple places where one could do this and this can easily lead to conflicting ideas of which features to use) leading up to the infamous global feature problem where all crates used in an application have to use the exact same features (cf. https://github.com/rust-embedded/cortex-m-rt/pull/107) even in different versions so each addition of new a new feature is potentially also a breaking change.

I'd really want to avoid a `cfg` gate hell of our own. The only alternatives I see here are:

- naming the new trait differently and slowly phasing out the old one
- simply replace it and release a new breaking version

ryankurte, 2018-10-11:
Good points ^_^ I kindof figured breaking changes should be few and far between, and our goal would be to keep most of our environment in sync with the hal, but the rand crate experience seems not great.

> In light of this, I now prefer the following approach:
> - Add new trait variations under a new name (e.g. MyTrait2) deprecate the old trait (e.g. MyTrait) and publish a non-breaking release.
> - On the [...]

I like that we'd be giving notice to consumers of depreciation, I don't like that we'd effectively be causing more breakage and thus fragmentation of dependencies.

> That means, a HAL implementation can release a transition version that supports both and old and a new version, allowing application authors to use any driver during the transition period.

It seems to me that if there's a way we can do the work on our side it'd be better than expecting it of the whole ecosystem? 

I just had [a go](https://github.com/rust-embedded/embedded-hal/blob/aacf7277c348b89e2972dabe34d125b03f2723a8/src/digital.rs) at supporting both by including a previous hal version and writing adaptors (as suggested by @therealprof I think, but I may have misunderstood the suggestion), but, it seems like it would become a bit of a nightmare to manage if we ever had overlapping breakages / dependency requirements.

> naming the new trait differently and slowly phasing out the old one

How would you approach this? I don't love the idea of version numbers in the api.

> simply replace it and release a new breaking version

Pretty onboard with this, we have a common understanding with semver of what a version number change means, it's simple for us and easy to understand, and it only breaks once.

hannobraun, 2018-10-12:
@ryankurte 
> It seems to me that if there's a way we can do the work on our side it'd be better than expecting it of the whole ecosystem?

If that's feasible, then we should do it. As an author/co-maintainer of several HAL implementations, I'm glad to have that in my toolbox though.

> > simply replace it and release a new breaking version
> 
> Pretty onboard with this, we have a common understanding with semver of what a version number change means, it's simple for us and easy to understand, and it only breaks once.

So we're back to the simple solution? I'm not opposed to this, but I think for a foundational crate like `embedded-hal`, it's worth exploring alternatives.

I see your points regarding version numbers in traits and the additional breakage, but I think both problems are not that important:
- The version numbers in the traits (e.g. `InputPin2`) are only a temporary measure. Temporary ugliness is okay in my book, if it serves a purpose (the purpose being a smooth transition period).
- The additional breakage causes minimal effort (e.g. change `InputPin2` to `InputPin`), and the breakage as a whole is spaced out over multiple releases. If we follow this transition strategy, every single one of those breaking releases will be such a non-issue, that the additional breakage is of no consequence, in my view.

Another point to consider:
Right now we can release breaking changes whenever it suits us. Once 1.0 is out, we should be much more conservative about breaking changes. That means, we'll either have a transition strategy in place by then, or we'll potentially be shipping broken traits with no alternative for what could be years. I think even the ugliest transition strategy (`InputPin2`, `InputPin3`, `InputPin4`, ... :) ) is preferable to that scenario.

Still, I'd be fine with clean, breaking releases, if that's the majority consensus. But I do think it's not the best solution, and not even a good solution in a post-1.0 world.

Suggestion:
Regardless of what we decide here for the general case, let's go with a clean break for #97. That way, we can take the time to come to a good decision without blocking that pull request.

ryankurte, 2018-10-14:
I definitely agree that we need to be more cautious for 1.0, and as mentioned above I think we should try to have whatever approach we decide on in place and tested before we get there.

+1 for a clean break for #97 so we can land it and move forward ^_^

hannobraun, 2018-10-17:
Relevant discussion in #92:
- https://github.com/rust-embedded/embedded-hal/issues/92#issuecomment-430636833
- https://github.com/rust-embedded/embedded-hal/issues/92#issuecomment-430658564

hannobraun, 2018-10-18:
@ryankurte 
> Afaik the same trait in different crate version is incompatible, so it seems to me that having a larger set of versions and traits over which we apply the changes is going to cause the same headache as my proposal with combinations of versions across dependencies instead of features.

Wouldn't this problem be solved by employing [the semver trick](https://github.com/dtolnay/semver-trick)?

> It sucks to have that impact, though should we go with the non-breaking merge and rename approach it would then require you to fix everything twice, and break all your downstreams twice? It seems to me that is also likely to leave people behind.

I really don't see this danger of leaving people behind.
- We presumably wouldn't release a breaking version with that one change, and instead wait until we have enough breaking changes to justify a breaking release.
- The changes will be minimal: Changing `InputPin2` to `InputPin` is no big deal, especially if you're already upgrading to a new version anyway.
- No single release will ever break anything. `InputPin2` will still exist, it will just be deprecated.
- We're not in a hurry. We can keep deprecated stuff around for as long as we want, until we're absolutely convinced the ecosystem won't be harmed by removing them. There's barely any harm in keeping `InputPin2` around forever, if that's what we decide is best.

Any crate that's "left behind" under these circumstances is essentially abandoned, and unless you're going to maintain all of those crates, that's nothing we can prevent :)

@therealprof 
> Yeah, not to happy about that either. I'd prefer to simply give the new traits fresh names and let the old names fade away into irrelevance and then remove them. We have a nice `#[deprecated]` flag to tell the users how to use the new traits and it's also rather simple for existing HAL implementations to provide both old and new trait impls.

I understand where you're coming from but I'd really hate if in 5 years, I have to type `FallibleInputPin` instead of `InputPin`, due to some old version of the trait that no one even remembers any more.

Yes, let's be super careful and super conservative about making these changes. Yes, let's keep those deprecated traits around until they have faded into irrelevance. But please, let's not lock us into cumbersome trait names forever, due to some stupid mistakes we've made in what's basically the stone age of the embedded Rust ecosystem.

Whether we call it `InputPin2` or `FallibleInputPin` during the transition phase is not my primary concern[1]. But let's at least intend to design a system that ends up at a near-optimal state.

[1] Although I prefer `InputPin2`, despite it's ugliness, because it communicates better that it's a new and improved version of `InputPin`, while `FallibleInputPin` doesn't, and is thus more likely to confuse casual observers.

therealprof, 2018-10-18:
> I understand where you're coming from but I'd really hate if in 5 years, I have to type FallibleInputPin instead of InputPin, due to some old version of the trait that no one even remembers any more.

Well, my 2 cents here:
- It's up to us to come with a more clever naming
- How often have you actually typed any of the trait names? You'll need that maybe a few times in an HAL impl, twice per driver impl using it and that's pretty much it...

I for one couldn't any less whether we call it `SuperDuperInputPin`, `InputPin2` or put put them into a new module with the same name or create a sub-module under digital with the same name or any combination of those.

ryankurte, 2018-10-19:
So I wrote a long reply without a good conclusion and decided to try implementing the semver trick inspired no-breakage update I mentioned earlier:

- [ryankurte/embedded-hal@v0.3.0](https://github.com/ryankurte/embedded-hal/tree/v0.3.0) for a clean break 0.3 release
- [ryankurte/embedded-hal/tree/v0.2.3](https://github.com/ryankurte/embedded-hal/tree/v0.2.3) backports the new traits to the old hal (see [digital.rs](https://github.com/ryankurte/embedded-hal/blob/v0.2.3/src/digital.rs) and [cargo.toml](https://github.com/ryankurte/embedded-hal/blob/2479c003765b80f2327dbcadac62e35e39788b79/Cargo.toml#L23))
- [ryankurte/hal-interop-example](https://github.com/ryankurte/hal-interop-example) demonstrates no-overhead interop between hal traits (see [lib.rs](https://github.com/ryankurte/hal-interop-example/blob/master/src/lib.rs)).

Which I think gives us:
- Compatibility between version
- No duplicated traits / forever numbering / multiple fixes required
- No changes required to resolve breakage (0.2.x minor update adds 0.3 compatibility)
- Nothing particular for us to maintain going forwards (all compatibility is in 0.2.x branch)
- We can still mark the 0.2 traits deprecated to move people forwards

I couldn't work out how to adapt the `StatefulOutputPin` trait in one direction, I think it should be possible but am not enough of a wizard with rust syntax to work out how. Apart from that, maybe I'm missing something, but it seems pretty good?

therealprof, 2018-10-19:
I gave v0.2.3 a quick spin and for some reason it breaks I2C for me:
```
cargo build --release --examples
    Updating git repository `https://github.com/ryankurte/embedded-hal`
    Updating git repository `https://github.com/ryankurte/embedded-hal`
   Compiling embedded-hal v0.3.0 (https://github.com/ryankurte/embedded-hal?tag=v0.3.0#f22caf24)
   Compiling embedded-hal v0.2.2 (https://github.com/ryankurte/embedded-hal?tag=v0.2.3#2479c003)
   Compiling stm32f042-hal v0.6.0 (/Users/egger/OSS/stm32f042-hal)
error[E0277]: the trait bound `hal::i2c::I2c<hal::stm32f0x2::I2C1, (hal::gpio::gpiof::PF1<hal::gpio::Alternate<hal::gpio::AF1>>, hal::gpio::gpiof::PF0<hal::gpio::Alternate<hal::gpio::AF1>>)>: embedded_hal::blocking::i2c::WriteRead` is not satisfied
  --> examples/i2c_hal_ina260serial.rs:55:26
   |
55 |         let mut ina260 = INA260::new(i2c, 0x40).unwrap();
   |                          ^^^^^^^^^^^ the trait `embedded_hal::blocking::i2c::WriteRead` is not implemented for `hal::i2c::I2c<hal::stm32f0x2::I2C1, (hal::gpio::gpiof::PF1<hal::gpio::Alternate<hal::gpio::AF1>>, hal::gpio::gpiof::PF0<hal::gpio::Alternate<hal::gpio::AF1>>)>`
   |
   = note: required by `<ina260::INA260<I2C>>::new`
```

therealprof, 2018-10-19:
I also breaks I2C in other crates with other drivers, e.g. SSD1306.

ryankurte, 2018-10-19:
Huh, how strange!

I based my 0.2.x branch on master, which does have [plenty of changes](https://github.com/rust-embedded/embedded-hal/compare/v0.2.1...master?expand=1), but I can't see any that should break I2C. I just released `v0.2.4` [with only my changes](https://github.com/ryankurte/embedded-hal/compare/v0.2.1...v0.2.4?expand=1), does that still break I2C? (And is there a project I can use to repro?)

therealprof, 2018-10-19:
Yes, same problem. I'm using my `stm32f042-hal` for this:

```
# git status
On branch master
Your branch is up to date with 'origin/master'.
# cargo build --release --examples
    Finished release [optimized + debuginfo] target(s) in 0.10s
# git stash pop
On branch master
Your branch is up to date with 'origin/master'.

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)

	modified:   Cargo.toml
# git diff
diff --git a/Cargo.toml b/Cargo.toml
index c12c7f5..0a6ce86 100644
--- a/Cargo.toml
+++ b/Cargo.toml
@@ -30,7 +30,9 @@ version = "0.2.2"

 [dependencies.embedded-hal]
 features = ["unproven"]
-version = "0.2.1"
+#version = "0.2.1"
+git = "https://github.com/ryankurte/embedded-hal"
+tag = "v0.2.4"

 [dependencies.stm32f0]
 features = [
# cargo build --release --examples
   Compiling stm32f042-hal v0.6.0 (/Users/egger/OSS/stm32f042-hal)
error[E0277]: the trait bound `hal::i2c::I2c<hal::stm32f0x2::I2C1, (hal::gpio::gpiof::PF1<hal::gpio::Alternate<hal::gpio::AF1>>, hal::gpio::gpiof::PF0<hal::gpio::Alternate<hal::gpio::AF1>>)>: embedded_hal::blocking::i2c::WriteRead` is not satisfied
  --> examples/i2c_hal_ina260serial.rs:55:26
   |
55 |         let mut ina260 = INA260::new(i2c, 0x40).unwrap();
   |                          ^^^^^^^^^^^ the trait `embedded_hal::blocking::i2c::WriteRead` is not implemented for `hal::i2c::I2c<hal::stm32f0x2::I2C1, (hal::gpio::gpiof::PF1<hal::gpio::Alternate<hal::gpio::AF1>>, hal::gpio::gpiof::PF0<hal::gpio::Alternate<hal::gpio::AF1>>)>`
   |
   = note: required by `<ina260::INA260<I2C>>::new`
...
```

ryankurte, 2018-10-19:
Right, it seems the fault occurs as soon as you add embedded-hal `v0.3.0` as a dependency to `v0.2.1`, so it isn't related to the traits, and renaming the dependency also doesn't help. I have no idea why this might happen :-/

ryankurte, 2018-10-22:
Hmm, sorry I thought I had posted an update but apparently it got lost in the ether :-/

It looks like the issue is that the `ina260` crate also includes `embedded-hal` at `v0.2.1`.  Updating this also to `v0.2.4` fixes the problem (though as I discovered the same version from different sources is considered a different crate).

Current:
```
➜  stm32f042-hal git:(master) cargo tree
stm32f042-hal v0.6.0 (file:///Users/ryan/projects/rust/hal-test/stm32f042-hal)
...
├── embedded-hal v0.2.1
│   ├── nb v0.1.1
│   └── void v1.0.2
...
[dev-dependencies]
├── ina260 v0.2.2
│   ├── cast v0.2.2 (*)
│   └── embedded-hal v0.2.1 (*)
```
Fixed:
```
➜  stm32f042-hal git:(master) ✗ cargo tree
stm32f042-hal v0.6.0 (file:///Users/ryan/projects/rust/hal-test/stm32f042-hal)
...
├── embedded-hal v0.2.4 (https://github.com/ryankurte/embedded-hal?tag=v0.2.4#0f37ad7b)
│   ├── embedded-hal v0.3.0 (https://github.com/ryankurte/embedded-hal?tag=v0.3.0#f22caf24)
│   │   ├── nb v0.1.1
│   │   └── void v1.0.2
...
[dev-dependencies]
├── ina260 v0.2.2 (file:///Users/ryan/projects/rust/hal-test/ina260)
│   ├── cast v0.2.2 (*)
│   └── embedded-hal v0.2.4 (https://github.com/ryankurte/embedded-hal?tag=v0.2.4#0f37ad7b) (*)
```
Build:
```
➜  stm32f042-hal git:(master) ✗ cargo build --release --examples                                                                                                   [6:21]
    Updating git repository `https://github.com/ryankurte/embedded-hal`
    Updating git repository `https://github.com/ryankurte/embedded-hal`
    Finished release [optimized + debuginfo] target(s) in 1.90s
```

We can mitigate this kind of issue with a note in the readme about how to troubleshoot it (and by using less-strict version matching, though that's not always appropriate), but afaik it's unavoidable with cargo's strict approach and happens with any (breaking or non-breaking) release. There's also an open issue to improve the errors emitted in this case https://github.com/rust-lang/rust/issues/22750.

therealprof, 2018-10-22:
A breaking change is a breaking change, no matter how well it is documented. The breakage may be less severe than simply changing the traits to something completely different but it still requires a new minor version to not cause a lot of pain for existing applications.

ryankurte, 2018-10-22:
Sorry for the dupe, I rewrote it before I discovered that gh was struggling.

> A breaking change is a breaking change, no matter how well it is documented. The breakage may be less severe than simply changing the traits to something completely different but it still requires a new minor version to not cause a lot of pain for existing applications.

I'm sorry, I agree and am not sure what you're raising here? My suggestion is to release a minor version update containing the updated APIs as a breaking change, and the backport for compatibility is no different than any other patch release?

therealprof, 2018-10-22:
> I'm sorry, I agree and am not sure what you're raising here? My suggestion is to release a minor version update containing the updated APIs as a breaking change, and the backport for compatibility is no different than any other patch release?

I don't think I understand: you'd like to have two minor versions, one for the change and one for the backport?

I'm still of the opinion that unnecessary breakage should be avoided at any cost and the mentioned changes are not significant enough to warrant such a sledge hammer.

ryankurte, 2018-10-23:
Almost the same as [above](https://github.com/rust-embedded/embedded-hal/issues/100#issuecomment-431225219), a new `v0.x.0` minor release (`v0.3.0`) would include the breaking IO trait changes, and a new `v0.2.x` patch release (`v0.2.2`) would introduce the compatibility backport. This leaves us with a new release with fallible traits and backwards compatibility with the `^0.2.2` line, letting people choose to either bump minor versions with no change or major versions and update their use of the traits, and either option interoperate.

This approach probably won't be applicable to all breaking changes, but it seems like a pretty reasonable mitigation for this one imo. It'll also be interesting to track what people do, given the options.

> I'm still of the opinion that unnecessary breakage should be avoided at any cost and the mentioned changes are not significant enough to warrant such a sledge hammer.

Specifically what _breakage_ are you talking about here? There are a bunch of sub-optimal outcomes from making and not making changes, and afaik any action we take causes _breakage_ of some kind. No change leaves us with flawed traits, releasing _any_ new package version causes _breakage_ so long as packages refer to different crate versions, having both trait types creates _breakage_ with incompatibility between drivers and hals using one or the other, changing the api creates _breakage_ because people have to update their code to match. 

As far as I can see, this approach mitigates about as much of this as we can, while still letting us move forward. All this is built on the premise that we should have fallible pins, and that the should replace the infallible ones rather than maintaining two alternatives, which I thought was addressed in #97, if that is not the case perhaps we need a wider wg vote?

therealprof, 2018-10-23:
> letting people choose to either bump minor versions with no change or major versions and update their use of the traits, and either option interoperate.

Doesn't that only work with all crates use exactly the same version as per your assessment?

> Specifically what breakage are you talking about here? 

If all crates (likely not under the control of the user) need to be on the same (or later) version that'd be a breaking change! If you found a way to make e.g. the `ina260` driver work without having to upgrade it to a newer `embedded-hal` version I'd be interested to learn about it.

> No change leaves us with flawed traits,

I don't see it that way here. The traits we're talking about here are perfectly fine for 99,9% of all use cases (which is GPIOs provided by an MCU). I cold also argue the opposite way by saying the fallible traits suggest a level of control and feedback which is typically not available and thus misleading.

In another case the situation might be different. I don't think there's a *one solution fits all*.

> releasing any new package version causes breakage so long as packages refer to different crate versions, having both trait types creates breakage with incompatibility between drivers and hals using [...]

No, compatible versions are compatible versions. It is *very* common to have different versions of crates involved and that is usually just fine. The cases where it is not are the ones I take offence at, and this is such a case.

 > All this is built on the premise that we should have fallible pins, and that the should replace the infallible ones rather than maintaining two alternatives,

And I disagree. I'd rather have two sets of traits, provide a bridge between them, declare the old ones deprecated and then simply phase out the infallible ones in a breaking version a while later. This leaves all crates sufficient time to adapt without causing **any** breakage at all.

> if that is not the case perhaps we need a wider wg vote?

A vote would be fine with me however I'd rather hear from other affected people how they feel about the proposed breakage.

hannobraun, 2018-10-23:
I have a question regarding the breakage that @therealprof experienced while testing @ryankurte's changes. If I understand correctly, the breakage was caused by the following circumstances:
- `ina260`'s dependency on `embedded-hal 0.2.x` came from crates.io
- `stm32f042-hal`'s dependency on `embedded-hal 0.2.x` came from @ryankurte's GitHub repository
- Therefore there were two versions of `embedded-hal 0.2.x` around. Those aren't the same crate, and therefore their traits aren't the same.

So, if my understanding is correct, then:
- This had nothing to do with with `embedded-hal 0.3.x`.
- This wouldn't be a problem, if we implemented this transition strategy in practice, as there never would be a second `0.2.x` version from GitHub.
- The correct way to test this would have been to use `[patch]` in the Cargo.toml.

Therefore it seems to me like @therealprof's argument against this approach is based on a wrong premise. Please correct me if I'm wrong, as I'm not sure I've completely understood everything that's going on.

@therealprof
> * How often have you actually typed any of the trait names? You'll need that maybe a few times in an HAL impl, twice per driver impl using it and that's pretty much it...
>
>
> I for one couldn't any [...]

It's not about typing the pin (even though that's what I said, I'm sorry :grin:). Short, concise names are valuable in understanding a system and making it accessible. Long, confusing names will continually confuse new  and infrequent users. Having an `InputPin` instead of a `SuperDuperInputPin` (or a `FallibleInputPin` for that matter) will keep paying dividends as long as embedded-hal is in use. Unless we expect the project to die in 1-2 years, I'd say that's worth more than some minor and alleviated breakage in the early days.

Plus, short and concise names are aesthetically pleasing. This won't win me the argument by itself, but it's a bonus point :)

@ryankurte
Quoting from https://github.com/ryankurte/embedded-hal/blob/v0.2.4/src/digital.rs#L21
``` rust
impl hal_v03::digital::OutputPin for OutputPin
```
I'm pretty sure this doesn't mean what you think it means :)
It implements the `hal_v03::digital::OutputPin` for trait objects (i.e. `&OutputPin`), not for types that already implement `OutputPin`.

I believe the correct way to do it is this:
``` rust
impl<T> hal_v03::digital::OutputPin for T where T: OutputPin
```
I'm not sure if that's going to run into problems with overlapping impls though.

---

Overall I'm still undecided, but cautiously optimistic about @ryankurte's proposal!

therealprof, 2018-10-23:
> Therefore it seems to me like @therealprof's argument against this approach is based on a wrong premise. Please correct me if I'm wrong, as I'm not sure I've completely understood everything that's going on.

You're wrong. 😉The problem is that one crate pulls in one version and the the other crate the other version and both define the traits differently; that does not work, independent of whether the crate comes from GitHub or does not.

> Plus, short and concise names are aesthetically pleasing. This won't win me the argument by itself, but it's a bonus point :)

Yeah, that doesn't say it's impossible to come up with a new concise name. More important than shortness for me is that a name is descriptive.

hannobraun, 2018-10-23:
> > Therefore it seems to me like @therealprof's argument against this approach is based on a wrong premise. Please correct me if I'm wrong, as I'm not sure I've completely understood everything that's going on.
> 
> You're wrong. winkThe problem is that one crate pulls in one version and the the other crate the other version and both define the traits differently; that does not work, independent of whether the crate comes from GitHub or does not.

Okay. Either you're wrong or I don't understand at all what's going on :)
I'll try to find the time to look into it more deeply. Maybe I'll find out something useful.

> > Plus, short and concise names are aesthetically pleasing. This won't win me the argument by itself, but it's a bonus point :)
> 
> Yeah, that doesn't say it's impossible to come up with a new concise name. More important than shortness for me is that a name is descriptive.

If we can come up with a new name that's actually better or just as good, that's the optimal scenario of course. And yes, descriptiveness is most important. I forgot to mention that explicitly.

ryankurte, 2018-10-25:
@therealprof 
> The traits we're talking about here are perfectly fine for 99,9% of all use cases (which is GPIOs provided by an MCU). I could also argue the opposite way by saying the fallible traits suggest a level of control and feedback which is typically not available and thus misleading.

Except for when they're not. We can choose the generic option that covers both cases and makes everything compatible, or split the api and create fragmentation / incompatibility between crates that support infallible or fallible IO. Providing traits to explicitly or implicitly cast between either puts the onus on users to manage our indecision, or hides the possibility of failure, neither of which is useful, and imo the split traits approach is undermines the whole objective of the hal in providing _useful_ interoperability.

As to the second part, this is usefully expressed with the [never type](https://github.com/rust-lang/rfcs/blob/master/text/1216-bang-type.md) (`type Error = !`) which is literally designed for this use case, once it's stabilised we can document that as a recommendation for infallible implementations.

@hannobraun 
> I have a question regarding the breakage that @therealprof experienced while testing @ryankurte's changes. If I understand correctly, the breakage was caused by the following circumstances ...

> Therefore there were two versions of embedded-hal 0.2.x around. Those aren't the same crate, and therefore their traits aren't the same.

Close but not exactly, both `ina260` and `stm32f042-hal` had **strict** dependencies on `embedded-hal` at **exactly** version `0.2.1`, updating only `stm32f042-hal` to `0.2.4` without changing `ina260` created explicit dependencies on two different crate versions, completely unrelated to any source code changes I introduced.

@therealprof 
> You're wrong. 😉The problem is that one crate pulls in one version and the the other crate the other version and both define the traits differently; that does not work, independent of whether the crate comes from GitHub or does not.

That is not correct, the traits in both 0.2.1 and 0.2.4 are the same. This issue is because **identical traits from different crate versions or sources are explicitly incompatible** (see all the issues linked from: https://github.com/rust-lang/rust/issues/53795, the error you demonstrated, and the resolution to it).

The _breakage_ you experienced will happen with _every single time we make a release_ so long as you have two explicit versions defined in nested crates. It's possible *you* could mitigate this with fuzzy semver matching as I mentioned above (`^0.2.1` I think is what you're looking for?) so cargo can try to use matching version. 

@hannobraun 
> The correct way to test this would have been to use [patch] in the Cargo.toml.

Yep, probably, but I find the docs on [patch] incomprehensible / unfortunately missing in detail and have *no idea* how it works :-/

> I'm pretty sure this doesn't mean what you think it means :)

> It implements the hal_v03::digital::OutputPin for trait objects (i.e. &OutputPin), not for types that already implement OutputPin.

That's what I though initially, but then I wrote [this](https://github.com/ryankurte/hal-interop-example/blob/master/src/lib.rs) and it worked. Maybe I did something silly in the example implementation, but, that's how I usually use peripherals?

ryankurte, 2018-10-25:
There also appears to be a precedent for backporting support between breaking versions with [v0.1.3](https://github.com/rust-embedded/embedded-hal/commit/0472df3f096c435f3cce16d76e9ef938c2fb63df) using the semver trick / as with my proposal (though v0.1.3 backports everything unchanged as far as I can see, which perhaps is what we should be doing), maybe it'd be useful to have @japaric's experience here.

therealprof, 2018-10-25:
> Except for when they're not.

True. But I prefer things to work smoothly for the majority of the cases and make the special cases a bit more involved than the other way around.

>  Providing traits to explicitly or implicitly cast between either puts the onus on users to manage our indecision, or hides the possibility of failure, neither of which is useful, and imo the split traits approach is undermines the whole objective of the hal in providing useful interoperability.

No it does not. Providing a clear migration path and deprecation markers makes it easy to ensure that the ecosystem adapts accordingly. Causing hard breakage causes user needless user frustration which we would to avoid at **any cost**.

> Close but not exactly, both ina260 and stm32f042-hal had strict dependencies on embedded-hal at exactly version 0.2.1, updating only stm32f042-hal to 0.2.4 without changing ina260 created explicit [...]

The main problem seems to be that the compiler tags traits with their location and doesn't try to check whether they're identical. 😞Effectively the impls are implementing different traits.

>  It's possible you could mitigate this with fuzzy semver matching as I mentioned above (^0.2.1 I think is what you're looking for?) so cargo can try to use matching version.

That's a good idea and I will try that.

> There also appears to be a precedent for backporting support between breaking versions with v0.1.3 using the semver trick / as with my proposal (though v0.1.3 backports everything unchanged as far as I can see, which perhaps is what we should be doing)

Yes, certainly.

hannobraun, 2018-10-25:
@ryankurte 
> @hannobraun
>
>
>
> Close but not exactly, both `ina260` and `stm32f042-hal` had **strict** dependencies on `embedded-hal` at **exactly** version `0.2.1`, updating only `stm32f042-hal` to `0.2.4` without [...]

Thank you for the explanation. Can you point me to the place where these strict dependency definitions are?

I can find the following for `ina260` ([source](https://github.com/therealprof/ina260/blob/84d895e37ed7f41fad83d8fd93ac3f57e4bd89a8/Cargo.toml#L23)):
``` toml
[dependencies.embedded-hal]
features = ["unproven"]
version = "0.2.1"
```

The one for `stm32f042-hal` looks identical ([source](https://github.com/therealprof/stm32f042-hal/blob/02050a7f27f07fda7d213f1655269c5156b2c44c/Cargo.toml#L31)), but was modified for the test to switch to a Git repository ([source](https://github.com/rust-embedded/embedded-hal/issues/100#issuecomment-431275413)).

This is **not** a strict definition. `version = "0.2.1"` is equivalent to `version = "^0.2.1"`. A strict definition would be `version = "=0.2.1"`.

If one were `version = "0.2.1"`, and the other one were `version = "0.2.4"`, Cargo would resolve both of them to version `0.2.4`.

Replacing either one, but not the other, with a Git version will make them into different crates. This would cause incompatibility unconditionally, either if both were otherwise totally identical and no semver trick was involved.

So far I've only assessed this based on the code on GitHub and what the both of you said here. Until I've had a chance to test it out myself I won't outright say that I'm right and the both of you are wrong but, well, that's my working theory right now :grin:

> The _breakage_ you experienced will happen with _every single time we make a release_ so long as you have two explicit versions defined in nested crates. It's possible _you_ could mitigate this with [...]

As I explained above, this is wrong. Saying `version = "0.x.y"` means `version >= 0.x.y && version < 0.x+1.0`.

Strict definitions would look like `version = "=x.y.z". I'm not sure if there's any valid use case to do this, ever. In any case, if you did this, you'd experience breakage all the time due to patch version upgrades. Protecting this use case from breakage must be out of scope for this discussion because, as far as I can see, the only way to protect this use case from breakage is not to release a new version ever again.

> @hannobraun
> 
> > The correct way to test this would have been to use [patch] in the Cargo.toml.
> 
> Yep, probably, but I find the docs on [patch] incomprehensible / unfortunately missing in detail and have _no idea_ how it works :-/

Something like this should do it:
``` toml
[patch.crates-io]
embedded-hal = { git = "https://github.com/path/to/repo" }
```

> That's what I though initially, but then I wrote [this](https://github.com/ryankurte/hal-interop-example/blob/master/src/lib.rs) and it worked. Maybe I did something silly in the example [...]

Are you sure? Because this is what I'm getting, after a fresh clone:
```
$ cargo +nightly test
   Compiling hal-backport-example v0.1.0 (/home/hanno/tmp/projects/hal-interop-example)                                                           
error[E0463]: can't find crate for `hal_v2`                                                                                                       
 --> src/lib.rs:2:1                                                                                                                               
  |                                                                                                                                               
2 | extern crate hal_v2;                                                                                                                          
  | ^^^^^^^^^^^^^^^^^^^^ can't find crate                                                                                                         
                                                                                                                                                  
error[E0463]: can't find crate for `hal_v2`                                                                                                       
 --> src/lib.rs:2:1                                                                                                                               
  |                                                                                                                                               
2 | extern crate hal_v2;                                                                                                                          
  | ^^^^^^^^^^^^^^^^^^^^ can't find crate                                                                                                         
                                                                                                                                                  
error: aborting due to previous error                                                                                                             
                                                                                                                                                  
For more information about this error, try `rustc --explain E0463`.                                                                               
[... cut ...]
warning: build failed, waiting for other jobs to finish...
error: Could not compile `hal-backport-example`.                                                                                                  

To learn more, run the command again with --verbose.
```

This error is not related to the blanked impls, but it makes me think you didn't actually test the code that's on GitHub right now (not saying you willfully didn't test it, I'm saying you must have been accidentally testing something else).

ryankurte, 2018-10-25:
> This is not a strict definition. version = "0.2.1" is equivalent to version = "^0.2.1". A strict definition would be version = "=0.2.1".

Oh gosh, [you are totally right](https://doc.rust-lang.org/cargo/reference/specifying-dependencies.html),  my bad. I am not used to caret being inferred / not explicit.

> but, well, that's my working theory right now 😁

Seems like a good working theory tbh.

> it makes me think you didn't actually test the code that's on GitHub right now (not saying you willfully didn't test it, I'm saying you must have been accidentally testing something else)

Fair concern, seems fine from my side though and a fresh clone works for me (on `cargo 1.31.0-nightly (5dbac9888 2018-10-08)`) :-/

```
➜  example git:(master) git status                                               [1:24]
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
➜  example git:(master) git remote -v                                          [1:24]
origin	git@github.com:ryankurte/hal-interop-example.git (fetch)
origin	git@github.com:ryankurte/hal-interop-example.git (push)
➜  example git:(master) ✗ git  push                                             [1:26]
Everything up-to-date
➜  example git:(master) ✗ cargo +nightly test                            [1:26]
    Finished dev [unoptimized + debuginfo] target(s) in 0.06s
     Running target/debug/deps/hal_backport_example-0168dd1c99b77478

running 4 tests
test tests::new_new ... ok
test tests::old_new ... ok
test tests::new_old ... ok
test tests::old_old ... ok

test result: ok. 4 passed; 0 failed; 0 ignored; 0 measured; 0 filtered out

   Doc-tests hal-backport-example

running 0 tests

test result: ok. 0 passed; 0 failed; 0 ignored; 0 measured; 0 filtered out
```

therealprof, 2018-10-25:
> So far I've only assessed this based on the code on GitHub and what the both of you said here. Until I've had a chance to test it out myself I won't outright say that I'm right and the both of you are wrong but, well, that's my working theory right now 😁

As I've said before: If one is using semantic versioning correctly then just incrementing the patch version shouldn't have any impact at all; those are compatible versions and will be treated as such by cargo. Fiddling with the traits makes the version incompatible, as can be observed by running `cargo semver` on the `v0.2.4` tag in @ryankurte version:
```
error: breaking changes in `OutputPin`
  --> /Users/egger/.cargo/git/checkouts/embedded-hal-68ea237a42de07e4/f22caf2/src/digital.rs:4:1
   |
4  | / pub trait OutputPin {
5  | |     /// Error type
6  | |     type Error;
7  | |
...  |
18 | |     fn set_high(&mut self) -> Result<(), Self::Error>;
19 | | }
   | |_^
   |
warning: added item to trait (breaking)
  --> /Users/egger/.cargo/git/checkouts/embedded-hal-68ea237a42de07e4/f22caf2/src/digital.rs:6:5
   |
6  |     type Error;
   |     ^^^^^^^^^^^

error: breaking changes in `set_low`
  --> /Users/egger/.cargo/git/checkouts/embedded-hal-68ea237a42de07e4/f22caf2/src/digital.rs:12:5
   |
12 |     fn set_low(&mut self) -> Result<(), Self::Error>;
   |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   |
   = warning: type error: expected (), found enum `std::result::Result` (breaking)

error: breaking changes in `set_high`
  --> /Users/egger/.cargo/git/checkouts/embedded-hal-68ea237a42de07e4/f22caf2/src/digital.rs:18:5
   |
18 |     fn set_high(&mut self) -> Result<(), Self::Error>;
   |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   |
   = warning: type error: expected (), found enum `std::result::Result` (breaking)
```

You might be totally right about the I2C problem vanishing into thin air by using a released crate instead of pointing to a git (or in my case local) repository, so I'm not worried about that at all. If it still breaks after a release we'll have to address this properly...

I **am** still wary and worried about the trait reuse for GPIO. The `0.2.0` breakage was a somewhat different case and did not have such a big impact as this will have.

hannobraun, 2018-10-25:
@ryankurte 
> Fair concern, seems fine from my side though and a fresh clone works for me (on `cargo 1.31.0-nightly (5dbac9888 2018-10-08)`) :-/

Sorry, that was my bad. Whatever the problem was, it's gone after `rustup update`. Now I'm confused why it works at all, but that's my problem, not yours ;)

@therealprof 
> As I've said before: If one is using semantic versioning correctly then just incrementing the patch version shouldn't have any impact at all; those are compatible versions and will be treated as [...]

But that's just it, what @ryankurte did really shouldn't have any impact! I can't see any reason why it wouldn't be a correct use of semver.

The `cargo semver` output you posted doesn't make sense to me. @ryankurte `v0.2.4` tag doesn't include any changes to those traits ([diff](https://github.com/rust-embedded/embedded-hal/compare/master...ryankurte:v0.2.4)).

It looks rather like `cargo semver` was run against `v0.3.0` ([diff](https://github.com/rust-embedded/embedded-hal/compare/master...ryankurte:v0.3.0)).

> I **am** still wary and worried about the trait reuse for GPIO. The `0.2.0` breakage was a somewhat different case and did not have such a big impact as this will have.

I'm not ruling out the possibility that I'm still missing something big, but at this point, it seems pretty clear to me that your objections are based on misunderstandings.

It seems to me that:
- Ryan's approach follows semver strictly.
- Ryan's approach allows for a smooth transition without forced breakage at any point.
- Ryan's approach seems to allow mixing and matching between old and new implementations, old and new drivers.
- Ryan's approach might even allow implementations to stay on old versions of `embedded-hal` indefinitely, and **still** implement traits from new versions, if we add a `pub` before `extern crate embedded_hal as hal_v03.

I still have some reservations about the nature of the blanket impl's, and would like to see more extensive testing. But I don't see any big conceptual problems with the approach at this point.

ryankurte, 2018-10-25:
@therealprof 
> As I've said before: If one is using semantic versioning correctly then just incrementing the patch version shouldn't have any impact at all

Yeah this was totally a misunderstanding of cargo semver on my part sorry, there have definitely been some on this side too 😅

@hannobraun 
> Ryan's approach might even allow implementations to stay on old versions of embedded-hal indefinitely, and still implement traits from new versions, if we add a pub before `extern crate embedded_hal as hal_v03.

If we base things on the previous breaking change we should replace any identical traits with the v03 ones, then maybe we don't need to export the whole v03 hal, just the new bits? idk. Also thanks for clearing up all my misunderstandings here ^_^

therealprof, 2018-10-25:
@hannobraun 

> The cargo semver output you posted doesn't make sense to me. @ryankurte v0.2.4 tag doesn't include any changes to those traits (diff).

No it wasn't. Maybe `cargo semver` is easily confused by the cross `impl`s.

>  it seems pretty clear to me that your objections are based on misunderstandings.

I would boldly claim that my picture is very clear. This proposal (reusing the trait names) is in the black magic realm and that kind of magic tends to fail in mysterious and sometimes spectacular ways which is exactly why I don't like it.

hannobraun, 2018-10-25:
@ryankurte 
> If we base things on the previous breaking change we should replace any identical traits with the v03 ones, then maybe we don't need to re-export the whole v03 hal, just the new bits? idk.

Yes, we absolutely need to replace identical traits with re-exports, otherwise the whole semver trick won't work. How we re-export additions to the API is an implementation detail, I think, and not really important in the grand scheme of things. I don't have a preference right now (although a single `pub use` would be less work).

> Also thanks for clearing up all my misunderstandings here ^_^

No problem, always glad if I can help :)

@therealprof 
> I would boldly claim that my picture is very clear. This proposal (reusing the trait names) is in the black magic realm and that kind of magic tends to fail in mysterious and sometimes spectacular ways which is exactly why I don't like it.

I am baffled by this comment.

First, making breaking changes to APIs without changing the names of those APIs is a practice that is absolutely pervasive and likely as old as the concept of APIs. The only novel thing here is the proposed approach to alleviating the breakage (which I wouldn't call reusing the name; bridging between the new and old versions seems more precise).

Second, you saying your picture is clear, then calling the proposal "black magic" seems to be a contradiction. Isn't "black magic" an expression used for things that are not understood or understandable? The only coherent interpretation of this comment that I can come up with is that you don't fully understand the proposal, don't think it can be fully understood, and therefore it's very clear to you we shouldn't use it.

I don't think that this is a very useful stance and I don't think this proposal is black magic. I think it is an interesting approach that can and will be understood. That understanding might unearth reasons why it can't work in practice, therefore my call for further testing.

therealprof, 2018-10-25:
@hannobraun 

> First, making breaking changes to APIs without changing the names of those APIs is a practice that is absolutely pervasive and likely as old as the concept of APIs.

No, reasonable projects don't do that unless there's an **important** reason to do it (which there isn't IMHO). The way this is typically done is to introduce a new API and deprecate the old API, then phase it out.

> Second, you saying your picture is clear, then calling the proposal "black magic" seems to be a contradiction. Isn't "black magic" an expression used for things that are not understood or understandable?

It is clear to **me**, that doesn't mean it is clear to **everyone**. That's exactly the problem with black magic, the wizards-in-charge think it's a great idea at the time until someone or something changes their mind... If you don't like the term black magic, we can also call it a footgun. Also (as expressed several times) I'm concerned about the users, not the few participants of this discussion.

> I think it is an interesting approach that can and will be understood.

And I disagree and don't appreciate your misguided interpretations of my comments.

jamesmunns, 2018-10-25:
Misc notes as an idle watcher of this thread:

* Although cargo will update items to their newest version, if you still had a stale `Cargo.lock` around, the versions will be pinned. This could cause a dependency to keep the old version, while the main crate takes the new version. This may explain why "broken" stuff worked with either a clean checkout (no lockfile), or with an upgraded/changed version of Rust (which I think at least partially invalidates cached items, but I am not 100% on this).
    * The fix for this is either `cargo update`, or deleting the target dir and lockfile.

Also, again as an outsider here (and as someone who has admittedly not read all of the context), this discussion seems to be getting a little heated. We're all working (hard, and in free time), to get this right. Perhaps it might be better to take this to a lower latency format to discuss iteratively? e.g. IRC, Hangouts, call, etc?

hannobraun, 2018-10-25:
@therealprof 
> @hannobraun
>
>
> No, reasonable projects don't do that unless there's an **important** reason to do it (which there isn't IMHO). The way this is typically done is to introduce a new API and deprecate [...]

I'm not disagreeing with that. I was observing that the practice is pervasive, not that it is good or should be emulated. This was brought on by my misunderstanding of your comment.

> It is clear to **me**, that doesn't mean it is clear to **everyone**. That's exactly the problem with black magic, the wizards-in-charge think it's a great idea at the time until someone or [...]

I'm very sorry for misunderstanding your comment. I didn't mean to offend you.

That still leaves me very confused about what your misgivings are specifically.

In any case, I won't attempt to continue this part of the discussion for now. I think it's obvious that one of us is operating based on misunderstandings, and I'd like do my homework to make sure that it isn't me. (Please, don't take this previous sentence as any kind of attack. I'm very aware of the fact this it could be me who's wrong, and all I'm saying is I want to experiment more before I'm ready to continue here).

@jamesmunns 
> * Although cargo will update items to their newest version, if you still had a stale `Cargo.lock` around, the versions will be pinned. This could cause a dependency to keep the old version, while [...]

Thank you, I'll keep that in mind.

> Also, again as an outsider here (and as someone who has admittedly not read all of the context), this discussion seems to be getting a little heated. We're all working (hard, and in free time), to [...]

I agree that we should try our best to not let this discussion get (more) heated, and I'd be happy to have a call with anyone who's willing to have one. I think that real-time communication is very valuable for the social function it has (in so far as it can remind us that we're all in the same boat here). But I doubt that any form of low-latency communication will be useful for resolving the technical argument.

therealprof, 2018-10-25:
@hannobraun 

> I'm very sorry for misunderstanding your comment. I didn't mean to offend you.

No offence taken.

> That still leaves me very confused about what your misgivings are specifically.

I'm opposing the reuse of the same trait names for the fallible versions because I think there's nothing inherently wrong with the infallible versions that would warrant such a radical API change because there's a good chance that this will cause confusion and frustration down the road.

My preferred approach is to name the fallible traits differently, mark the old ones as deprecated, keep the fallible->infallible `impl` from @ryankurte and release the whole thing as quickly as possible a new patch or minor version (to signify the change even without breakage as recommended for >1.0.0 versions here: https://doc.rust-lang.org/cargo/reference/manifest.html#the-version-field).

therealprof, 2018-10-25:
Sounds good.

ryankurte, 2018-10-25:
> Before we can agree on an approach, we need to agree on a basic value framework first

I agree, I wonder whether it'd be useful to talk about goals/objectives/expectations for the embedded-hal somewhere else, though that discussion should hopefully not block us making changes, maybe a big issue that people can drop their perspectives into could be a way to start?

> Therefore I think it's worth putting effort into finding an approach that can potentially allow us to end up with a simpler end result, without causing pain right now.

I agree also. My perspective is that the hal exists to provide simple compatibility, and anything that can result in fragmentation of implementations is worth being wary of. There are a lot of people using the project now, hopefully in the future it will be orders of magnitude more, but it does depend on us getting this _right_.

> Given that, I'd like to propose the following. Short-term (i.e. right now), we do this ...

Sounds good to me.

hannobraun, 2018-10-28:
> I agree, I wonder whether it'd be useful to talk about goals/objectives/expectations for the embedded-hal somewhere else, though that discussion should hopefully not block us making changes, maybe a [...]

I agree that a such an issue sounds like a good way to start.

I think it would also be a good idea if someone wrote up the consensus that seems to have developed here (release breaking changes under new names, research other approaches, revisit later) and opened a pull request that adds this as guidelines to this repository. Then we can have a proper team vote on it.

hannobraun, 2018-11-08:
Hey, sorry for the delay. I'm away from the office right now, and this got buried under loads of unhandled notifications.

@ryankurte
> Assuming for now we're in agreement, @hannobraun you raised concerns about `impl hal_v03::digital::OutputPin for OutputPin` vs. `impl<T> hal_v03::digital::OutputPin for T where T: OutputPin` ([#100 (comment)](https://github.com/rust-embedded/embedded-hal/issues/100#issuecomment-432306664)), as well as with limiting the ability to implement both traits ([#100 (comment)](https://github.com/rust-embedded/embedded-hal/issues/100#issuecomment-433116884)). As far as I can tell (given we provide adaptors) we shouldn't need anyone to implement both trait versions?

Assuming the adapters work, I can't think of a reason why anyone would need to implement both trait versions. I just wanted to bring it up, in case someone else can think of a reason why this would be a problem.

> Do you have an alternative approach we could explore, and what testing would you like to see before we go ahead with one?

I don't have other ideas. If I were to go ahead with this personally (which I don't have time for right now), I'd want to experiment with `impl ... for OutputPin` vs `impl<T> ... for T ...`, to fully understand what the difference is, and why `for OutputPin` works (which surprises me). I don't have any specific test cases in mind.

Since I'm not going to do the work, and don't have concrete concerns, I won't object to going ahead with this.

ryankurte, 2019-05-13:
more discussion of how _cursed_ breaking updates are in #135 :-(

ryankurte, 2020-06-12:
Following the [hal team meeting](https://hackmd.io/ck-xRXtMTmKYXdK5bEh82A?view) and [updated process](https://github.com/rust-embedded/embedded-hal/pull/197) I think we can close this. 

Thank you all for your attention and effort!

closed 2020-06-12 by ryankurte