rust-embedded/embedded-hal #115: Enable use of generic peripheral creation

aurabindo, 2018-12-09:
The fact is, hardware is vendor dependent. If embedded-hal takes a firm position that we only include generic interfaces, then it would not likely end up being a go to solution for someone hoping to use a specific feature in the hardware which is almost always the case in a production level system. So rather than fully rejecting vendor specific stuff, I think we need to first agree on the ground rules on which we can disagree with vendor specific / hardcoded hardware features.

Is it possible to have a generic half of the embedded-hal and a vendor specific half? This way one can add features specific to a particular hardware in a way it will play nice with rest of the abstractions and would be interoperable. For example on Nordic chips, there is a feature "Programmable Peripheral Interconnect" which enables events on one peripheral to trigger tasks in other peripherals. There is no point in trying to generalize such hardware features. But embedded-hal should have mechanism to let such features be exposed to the end user in a safe API. Otherwise the user will be required to do unsafe code. HAL implementations are free to create an API around this peripheral. When everyone decides to do things their way, there would be fragmentation and inconsistency in the ecosystem. If embedded-hal could theoretically allow any kind of feature to be built by being least intrusive, it would a great addition to the ecosystem.

Coming to the PPI example, this would mean a module `embedded_hal_vendor::nordic::nrf52::nrf52832::ppi`. Within the module it contains helper functions to do various configurations and abstraction for registers. Rust language features & embedded-hal imposed conditions must be used to enforce various safety aspects. For example, there may be channels which have hardcoded conditions on source/target peripherals which should not be allowed.

Another missing piece in embedded-hal is support for interrupt configuration for various peripherals like I2C, SPI etc.

Is there any interest in the community for exploring such possibilities ?

ryankurte, 2018-12-11:
Hey, interesting suggestion. The embedded-hal is intended to provide common traits to sit between peripheral drivers and hardware implementations, for things like SPI that are available on a device and used to communicate with another device/peripheral, not to expose all functionality on a device, or device configuration.

You're right that we should have safe abstractions, but in this instance this would be a part of the [nrf-rs](https://github.com/nrf-rs) project as a device or family specific interface.

If it turns out that there are some driver uses for PPI, and we can generalise it across vendors, then it _may_ be a candidate for inclusion, but for now I do not believe that is the case.

closed 2018-12-11 by ryankurte

aurabindo, 2018-12-12:
@ryankurte Thanks for the response. We certainly cant have major architecture changes for embedded-hal, and if PPI is worth generalizing, then it would be nice too. It is consistent with the existing policy. I do understand that nrf-rs can have device specific additions which may be out of the embedded-hal's scope. I've summarized my intention in the wishlist https://github.com/rust-embedded/wg/issues/256#issuecomment-446665857