rust-embedded/embedded-hal #144: Proposal: SPI HalfDuplex and Device traits

no111u3, 2019-06-17:
Currently SPI have one model of work - Master + FullDuplex. But some mcu and mpu have more intelligence SPI implementation: HalfDuplex, variable length of transfer (4 - 16 bit), Device mode.
I thinks we need to add HalfDuplex and Device trait for current SPI interface. Also I think that adding of 8:16 bit transfer is not difficult.

dzarda, 2020-01-30:
IMHO half duplex will see most usage among features you mentioned.

ryankurte, 2020-01-30:
hey, thanks for opening an issue, a few questions and comments:

- what do you mean by half duplex? afaik this is covered by the[`blocking::spi::Write`](https://docs.rs/embedded-hal/0.2.3/embedded_hal/blocking/spi/trait.Write.html) trait.
- as far as variable length transfer goes, the SPI traits are currently generic over `W` allowing implementations to define supported word sizes (though in practice I haven't seen any non-`u8` implementations yet).  static definition of variable length transfer thus would involve adding implementations or bindings for the relevant word sizes, is this unsuitable for a particular application?
- Device mode certainly would be interesting! There's a related issue for I2C here https://github.com/rust-embedded/embedded-hal/issues/179

dzarda, 2020-01-30:
@ryankurte half duplex references a bidirectional mode where MISO and MOSI are shared (one signal). Master then generally needs to:
1. write multiple bytes (driving "MOSO")
2. read multiple bytes (sensing "MOSO")

Please see BIDIMODE and BIDIOE bits from STML0x3 manual (SPI control register 1):
[image]

therealprof, 2020-01-30:
@dzarda Sure, but why do you need special traits for that case? The traits only care about the data transfer, not the initialisation or the hardware connected to it. So unless I'm missing something here you only need to have a HAL impl that provides a half-duplex setup but the usage would be exactly the same.

ryankurte, 2020-01-31:
> @ryankurte half duplex references a bidirectional mode where MISO and MOSI are shared (one signal).

Ahh, that makes sense, thanks!

> So unless I'm missing something here you only need to have a HAL impl that provides a half-duplex setup but the usage would be exactly the same.

I guess we're missing a `Read` trait to support this?

hannobraun, 2020-01-31:
@ryankurte 
> the SPI traits are currently generic over `W` allowing implementations to define supported word sizes (though in practice I haven't seen any non-`u8` implementations yet).

Here: https://github.com/stm32-rs/stm32f7xx-hal/blob/master/src/spi.rs
Specifically: https://github.com/stm32-rs/stm32f7xx-hal/blob/af6900285648d48decb8ac5e0f8c74ef72e5ec4b/src/spi.rs#L738-L763

therealprof, 2020-01-31:
> I guess we're missing a Read trait to support this?

Not sure, you still have to clock out a number of `Word`s to do the `Read`s so I guess the question really boils down to whether this already possible in an efficient way now. If the answer is no, then we'll need a `Read` trait.

dzarda, 2020-01-31:
TBH I'm confused about the contract on send() and read(). Would these be unused in HalfDuplex? AFAIK each send() must be followed by read() otherwise some Overflow condition ensues.

dzarda, 2020-02-01:
-> Of course they wouldn't be used, they belong to the `FullDuplex` trait.

dzarda, 2020-02-02:
[blocking::spi::Write](https://docs.rs/embedded-hal/0.2.3/embedded_hal/blocking/spi/trait.Write.html) - sure, that could be used for sending data. But the receive functionality is simply absent. It's not a matter of efficiency AFAIK, I don't see a way of mapping this onto current API.

dzarda, 2020-02-10:
I evaluated the idea on the STM32L0x3 where the "bidirectional" receive behaviour is to output SCK clock with BIDIMODE=1 and BIDIOE=0. Not sure how to map this onto some API. More devices should be researched.

ierturk, 2023-11-10:
Is there any update?