rust-embedded/embedded-hal #152: Reading data from serial over bluetooth

JPFrancoia, 2019-09-30:
Hello,

I'm more or less new to Rust and embedded programming. I'm trying to read a TMP36 temperature sensor (it's analogic), and send the measurement with serial over bluetooth. I have a blue pill, a HC-05 bluetooth module, and...well, a TMP36.

Let's assume my wiring is ok for now. Here is the code I'm using:

```rust
// #![deny(unsafe_code)]
// #![deny(warnings)]
#![no_main]
#![no_std]

extern crate cortex_m as cm;
extern crate cortex_m_rt as rt;
extern crate panic_itm;
extern crate stm32f1xx_hal as hal;
extern crate embedded_hal;
extern crate nb;

use nb::block;

use hal::prelude::*;

use hal::{
    adc,
    serial::{Config, Serial},
    pac,
};

use embedded_hal::digital::v2::OutputPin;

use rt::entry;

#[entry]
fn main() -> ! {
    // Get control of the PC13 pin
    let device_peripherals = pac::Peripherals::take().unwrap();
    let cortex_peripherals = cortex_m::Peripherals::take().unwrap();

    let mut rcc = device_peripherals.RCC.constrain();
    let mut flash = device_peripherals.FLASH.constrain();
    let clocks = rcc.cfgr.freeze(&mut flash.acr);

    let mut gpioa = device_peripherals.GPIOA.split(&mut rcc.apb2);
    let mut gpiob = device_peripherals.GPIOB.split(&mut rcc.apb2);

    // Prepare the alternate function I/O registers
    let mut afio = device_peripherals.AFIO.constrain(&mut rcc.apb2);

    // USART1
    let tx = gpioa.pa9.into_alternate_push_pull(&mut gpioa.crh);
    let rx = gpioa.pa10;

    let serial = Serial::usart1(
        device_peripherals.USART1,
        (tx, rx),
        &mut afio.mapr,
        Config::default().baudrate(9600.bps()),
        clocks,
        &mut rcc.apb2,
    );

    // Setup ADC
    let mut adc1 = adc::Adc::adc1(device_peripherals.ADC1, &mut rcc.apb2, clocks);

    // Split the serial struct into a receiving and a transmitting part
    let (mut tx, _rx) = serial.split();

    // Init analog temperature sensor
    let mut tmp36 = gpiob.pb1.into_analog(&mut gpiob.crl);

    // We can only send byte per byte with the serial
    let data: u16 = adc1.read(&mut tmp36).unwrap();
    
    for byte in data.to_be_bytes().iter() {
        block!(tx.write(*byte)).ok();
    }

    loop {}
}
```

The code compiles, and I can send it to the blue pill. When I run the code line by line in the debugger, I can see that `data` (the raw measurement from the sensor) has a value. But `data` is of type `u16`.

The `write` function of the `tx` pin takes an `u8`. So I naively tried to split the u16 into an u8. I can see that some data is transmitted over serial (I have `screen` open, reading the serial port). But the only thing I see coming is: � . It obviously means I don't read the bytes properly, or that I'm sending improper data.

Do you have an idea about how I can send and read the data properly?

Cheers

burrbull, 2019-10-01:
`screen` try to parse your numbers as ascii symbols.
And why big endian?

Rahix, 2019-10-01:
You have to decide what you want to do: Either you want to send ASCII encoded text (or any other encoding for that matter), or you just want to send the raw byte data as you are doing now.

For viewing in `screen`, you need to send ASCII text, which means you will have to convert your number into a string.  You can, for example, use the [`itoa`](https://crates.io/crates/itoa) crate for this.

For an example, suppose your ADC returns the value 51966.  The raw (big endian) bytes for this would be `0xca 0xfe`, which is what you are sending now.  `screen` tries to interpret these numbers as encoded text and will fail to decode them, thus showing the `�` symbol.  Instead, you want to send the bytes `0x35 0x31 0x39 0x36 0x36 0x0d 0x0a` which decode to the string `"51966\r\n"`.  This is now something which can be properly displayed by `screen`.

Keep in mind that you have to use `\r\n` as line-endings as this is a raw tty device.

JPFrancoia, 2019-10-03:
Yes! Indeed I didn't think about converting to string first. This works with:

```rust
    // We can only send byte per byte with the serial
    let data: u16 = adc1.read(&mut tmp36).unwrap();

    let mut buffer = itoa::Buffer::new();
    let printed = buffer.format(data);

    for byte in printed.bytes() {
        while block!(tx.write(byte)).is_err(){};
    }
```

Isn't there an easier way to do this though? I mean, a function like "print_serial" or something like that?

burrbull, 2019-10-04:
With [this](https://github.com/stm32-rs/stm32f1xx-hal/pull/111) you can use:
```rust
writeln!(tx, "{:?}", data.to_be_bytes());
```

burrbull, 2019-10-04:
Or you can plot your data in real time with [SVisual](https://github.com/Tyill/SVisual/) and [this client](https://github.com/burrbull/svisual-stm32f1/).

JPFrancoia, 2019-10-06:
Ok thanks, I take note. I didn't use the `writeln!` for now, but I'm sure it will become handy once my projects get bigger. FYI, I put my code + a readme online:

https://github.com/JPFrancoia/blue_pill_tmp36

closed 2019-10-06 by JPFrancoia