rust-embedded/embedded-hal #156: Change reference implementation to something more maintained

TheZoq2, 2019-10-26:
While scrolling through the docs tooday, I noticed the line

> The [stm32f30x-hal](https://crates.io/crates/stm32f30x-hal/0.1.0) crate contains a reference implementation of this HAL.

As far as I can tell, this crate hasn't been updated in two years, and is kind of superseded by the stm32-rs maintained version.

Therefore I think it would make sense to switch the reference implementation to something else, or to remove the line completely.

therealprof, 2019-10-26:
@TheZoq2 Which docs? Can you do a PR with the fix?

TheZoq2, 2019-10-27:
I guess I should have clarified that :P. The main page of the docs on docs.rs https://docs.rs/embedded-hal/0.2.3/embedded_hal/#reference-implementation

I'll gladly do a PR, but i'm not sure what to replace it with. Is there still such a thing as a reference implementation? Perhaps we could link to the HAL section of awesome-embedded-rust instead

thejpster, 2019-10-28:
Good question. @jamesmunns can we stick this on the agenda for Tuesday? Personally, I think there should be a reference implementation - perhaps for lm3s811evb or lm3s6965evb emulated by QEMU.

jamesmunns, 2019-10-28:
@thejpster you probably want to ping @therealprof this week, I will likely not be there this week (same with the other issues you tagged).

no111u3, 2019-11-24:
@thejpster @jamesmunns 
> Good question. @jamesmunns can we stick this on the agenda for Tuesday? Personally, I think there should be a reference implementation - perhaps for lm3s811evb or lm3s6965evb emulated by QEMU.
I can rewrite this part of examples and remove not common code. Btw now I create new `clean` PR for embedded-hal. Thanks

rubberduck203, 2020-02-03:
I noticed this today as well. I went back and found the wg minutes.
This issue was mentioned, but there were no notes on the result of the conversation.
Does anyone remember the result?

https://github.com/rust-embedded/wg/blob/master/minutes/2019-10-29.md

TheZoq2, 2020-06-18:
Looks like this is even more wrong in the `1.0.0.alpha` version since the mentioned f3 crate is stuck at `0.2.0`

eldruin, 2020-06-18:
Linking to #177

eldruin, 2020-09-22:
Now the `stm32f3` is used directly. Closing.

eldruin, 2020-10-02:
The reference to `stm32f30x-hal` is still present in the docs. Reopening

commit 8d318939 (2020-10-06): Merge #251

251: Update reference implementation implementation to stm32f1xx-hal and update examples r=therealprof a=eldruin

I think the `stm32f1xx-hal` is a good candidate for a reference implementation.
I have also updated the examples accordingly and modernized them a bit.

An additional possibility would be to remove the `futures` examples as well, since async handling will be different.

Closes #156
PR #251 (2020-10-06): Update reference implementation implementation to stm32f1xx-hal and update examples
I think the `stm32f1xx-hal` is a good candidate for a reference implementation.
I have also updated the examples accordingly and modernized them a bit.

An additional possibility would be to remove the `futures` examples as well, since async handling will be different.

Closes #156
closed 2020-10-06 by bors[bot]