rust-embedded/embedded-hal #187: Video call scheduling

thejpster, 2020-02-25:
At the WG meeting it was agreed the HAL team would meet to thrash out a way forward for the HAL. This would include:

The agenda of that meeting to cover:

* How we organize embedded-hal (one crate vs many)
* How we "mainline" items
* What we do about the state of things (v2, v3) right now

I propose Thursday 27 Feb at 1900 Europe/London. VC details to follow.

@rust-embedded/hal please confirm whether you can or cannot make this time. Please also suggest further agenda items if required.

therealprof, 2020-02-25:
Works for me.

Disasm, 2020-02-25:
Works for me too

ryankurte, 2020-02-28:
oops, guess i missed this one. did it happen / did anyone take minutes?

therealprof, 2020-02-28:
@ryankurte That meeting did not happen, due to being too short on notice to arrange.
@jamesmunns Do you have any preferred date? I'll be travelling next week so that won't really work for me.

thejpster, 2020-02-29:
Oops. I proposed a time and totally forgot to follow up and set up a call. My bad.

Next week I've only got Tuesday evening free.

therealprof, 2020-03-01:
Next week I'm at a conference.

ryankurte, 2020-03-02:
How about on the 11th, directly after the WG meeting?

thejpster, 2020-03-03:
I can do 2020-03-11T20:00:00Z.

Disasm, 2020-03-03:
I believe, meeting will be on 10th, but this date works for me too

ryankurte, 2020-03-03:
> I believe, meeting will be on 10th, but this date works for me too

ahh yes, sorry, i intended that to be on the 10th @ 20:00 GMT but forgot to convert the date (i live in the GMT+13 future so this is a morning meeting for me 🙃)

:tada: for 2020-03-10T20:00Z
:rocket: for 2020-03-11T20:00Z

ryankurte, 2020-03-09:
So confirming availability on 2020-03-10T20:00Z (directly following the embedded WG meeting / ~22 hours from now)

- [x] @ryankurte 
- [x] @therealprof 
- [ ] @thejpster 
- [ ] @Disasm 

Once y'all have confirmed I'll ask someone (probably @jamesmunns) nicely to add a calendar event with a hangouts chat

thejpster, 2020-03-10:
> So confirming availability on 2020-03-10T20:00Z (directly following 
the embedded WG meeting / ~22 hours from now)

Works for me.

jamesmunns, 2020-03-10:
I can attend/set up a meeting. It'll be on Zoom.

jamesmunns, 2020-03-10:
The meeting details are:

https://zoom.us/j/894390824

Password 070334

ryankurte, 2020-03-10:
thanks team!

meeting notes: https://hackmd.io/ck-xRXtMTmKYXdK5bEh82A?view#Final-Summary

closed 2020-03-10 by ryankurte