rust-embedded/embedded-hal #188: I2c slave specific registers

Stupremee, 2020-03-03:
I'm currently implementing hal support for [libtegra](https://github.com/mirage-rs/libtegra).
While I tried to implement the i2c traits, it did not work because we have special registers for different slaves. The i2c traits only have the slave as argument but not the register. How can I implement this with embedded-hal? Our current I2c api looks like [this](https://github.com/mirage-rs/libtegra/blob/master/src/i2c/device.rs#L260)

david-sawatzke, 2020-03-03:
It seems to me like your I2C interface is a bit higher level than embedded-hal. `send_packet` & `receive_packet` are more in line with the embedded-hal i2c traits. 

I2C itself doesn't have any notion of registers, it's just a very common interface that I2C devices use.

Stupremee, 2020-03-03:
Ah okay. Thank you.

closed 2020-03-03 by Stupremee