rust-embedded/embedded-hal #190: Serial: Sending break condition

dbrgn, 2020-03-06:
It seems like an embedded-hal driver cannot issue a USART break condition using only the embedded-hal serial trait, right?

> A break condition is signaled to the module by keeping the UART_RX pin
> low for longer than the time to transmit a complete character. For example,
> at the default baud rate of 57600 bps keeping the UART_RX pin low for 938
> us is a valid break condition, whereas at 9600 bps this would be interpreted
> as a 0x00 character. Thus, the break condition needs to be long enough
> to still be interpreted as such at the baud rate that is currently in use.

It seems that a HAL would have to reconfigure the pin as GPIO output pin, pull it low for a certain duration (depending on baudrate), then reconfigure it as serial pin again.

Would it make sense to add a `BreakCondition` trait to the `serial` module?

```rust
trait BreakCondition {
    fn send_break_condition(self) -> Self;
}
```

(Taking `self` by value is probably necessary in case the HAL needs to temporarily reconfigure the pin.)

jonas-schievink, 2020-03-06:
IMO taking `self` by value to work around limited HALs is not a good idea. HALs should consider implementing something like this API instead: https://github.com/stm32-rs/stm32l0xx-hal/pull/74

dbrgn, 2020-03-06:
@jonas-schievink good call, I updated the proposal.

jonas-schievink, 2020-03-06:
Okay, that wouldn't work with a nonblocking API I suppose