rust-embedded/embedded-hal #203: I2S interface

maxekman, 2020-04-16:
First some background: I'm planning to use embedded Rust in a sound-oriented project and am currently experimenting with the STM32 F4 Discovery which has an onboard DAC with amp and headphone jack. The DAC connects to the MCU with I2C for control commands (volume etc) and I2S for the audio data. Currently I have just started to implement a embedded-hal driver for the DAC, a Cirrus Logic CS43L22, beginning with setting up the device and I2C control channel. In the end I will use a more capable DAC with 4-8 channels, but it will still use I2S for audio data.

Is there a plan to add a I2S interface in the embedde-hal project yet? If not, how would I go about contributing that to embedded-hal? I have read some guidelines where it says to create a new interface in an external crate before it's stable, any example of that process so far?

The I2S standard uses mostly the same internal components and control registers as a SPI interface, but with a bit more complex clock setup to my understanding.

ryankurte, 2020-04-16:
hi, sounds like an interesting project!

> Is there a plan to add a I2S interface in the embedded-hal project yet?

As far as I know there is not yet.

> I have read some guidelines where it says to create a new interface in an external crate before it's stable, any example of that process so far?

You can either create a new crate with new traits, or fork embedded-hal and add traits (and use cargo patches to test), we've just switched to this approach so don't have a great example of the former yet, though #191 shows the latter.

eldruin, 2020-04-17:
Interesting project. I also have several I2S devices (including a CS4344 :) I could write a driver for but I always focused on other projects.
To start the discussion I have created a PR at #204.

maxekman, 2020-04-17:
Thanks for all the input! I’m fairly new to Rust and will follow along the proposal and contribute what I can to the process.

eldruin, 2022-11-11:
I published I2S traits in a separate [`embedded-i2s`](https://github.com/eldruin/embedded-i2s-rs) crate.
I would encourage anybody to continue the discussion over there.
A discussion about integration into `embedded-hal` would come much later.