rust-embedded/embedded-hal #205: Unable to use `InputPin`

David-OConnor, 2020-04-18:
I'm not sure what's going on, but because I don't experience this issue with `OutputPin`, I speculate it's due to `InputPin` being uproven:

```rust
pub fn f<P: embedded_hal::v2::digitial::InputPin>(pin: &P)  {
    };
```
^^^^^^^^^^^^^^^^^^^^ the trait `embedded_hal::digital::v1::InputPin` is not implemented for `&rppal::gpio::pin::InputPin`

I'm not sure why `v1`'s in there. It's not implemented, but I don't understand what I did to require it to be implemented.

ryankurte, 2020-04-19:
hmm, strange. this looks like it could be related to https://github.com/rust-embedded/embedded-hal/pull/199 if `rppal` has not been updated to `v2` traits, which you could test with cargo patch / is waiting on me having time to put together a patch release.

eldruin, 2020-04-25:
`rppal` is using the `v2` traits only in the master branch. This was not released yet.
@David-OConnor can you try using the master branch version of `rppal`?

David-OConnor, 2020-04-25:
Thank you very much - it sounds like that's it. I'm still getting the same `v1` error, but haven't spent much time troubleshooting.

Is this how you specify the dep?
```toml
rppal = { git="https://github.com/golemparts/rppal", branch="master", features=["hal-unproven"] }
```

David-OConnor, 2020-04-25:
Working now using that git.

David-OConnor, 2020-05-02:
This seems to have something to due to using the structs directly, vice having them borrowed from a `RefCel<Rc<>>`. Ie the latter triggers it while the former doesn't. Still can't pin down where the `v1` requirement is coming from.

eldruin, 2020-07-15:
We made some fixes in `embedded-hal` `0.2.4`.
@David-OConnor Could you run `cargo update` and see if the problem is solved?