rust-embedded/embedded-hal #207: Datetime traits

David-OConnor, 2020-05-01:
What are your thoughts on `Datetime`, `Time`, `Date` etc traits? They seem like a good fit because they're not available natively on `no_std`, are a common feature. It would be nice to be able to write general functions using these traits, instead of platform-specific ones.

Ie, make it so you don't have to hard-code functions to accept `chrono::Datetime` (eg a computer) or `ds323x::DateTime` (eg `ds3231`).

eldruin, 2020-05-01:
The types from [`chrono`](https://crates.io/crates/chrono) can be used in `no_std` as well. I use them in the [`rtcc`](https://crates.io/crates/rtcc) crate where I define a trait for RTC devices. I implemented this in the `mcp794xx` driver but did not implement this yet in the `ds323x` or `ds1307` drivers.

David-OConnor, 2020-05-01:
That's awesome

eldruin, 2020-05-01:
I'm currently implementing the trait for the `ds323x` driver. Stay tuned.

David-OConnor, 2020-05-01:
Think I should PR `linux-embedded-hal` with an impl using Chrono?

David-OConnor, 2020-05-02:
Btw, looking fwd to your `Rtcc` impl on `ds32x`; I'm writing the code for it now using `ds323x`'s syntax. The obvious appeal of the `Rtcc` syntax is ability to switch devices without changing the code too much.

eldruin, 2020-05-02:
Ok, I just published version 0.3 of the [ds323x](https://crates.io/crates/ds323x) driver crate which implements the [`Rtcc`](https://crates.io/crates/rtcc) trait.
Yes, the ability to switch between implementations without changes to the code that just uses the traits is the main benefit from defining and implementing common traits. You can easily switch between using a DS3231, MCP79401 or any other RTC device supported by the drivers, for example.

I am unsure whether chrono-based methods for `linux-embedded-hal` would make sense. It would definitely make sense if we would add `Rtcc`-like traits to `embedded-hal`. Otherwise, if you are in a system running Linux, you can already use `chrono` and `std` methods anyway, I would say.

Whether `Rtcc`-like traits in `embedded-hal` would make sense, I do not know. RTCs are present in many systems but I would be a bit wary of adding the dependency on `chrono` to `embedded-hal`.
Maybe somebody else has an opinion here?

David-OConnor, 2020-05-02:
# 1: Switched to your new `ds323x` crate and the `Rtcc` trait - awesome! This is seriously good stuff, and is what I was looking for with this issue.

# 2: I've written most of the `linux-embedded-hal` thing (Get pulls things from `Chrono::Utc`, and set uses `process::Command` to set sys time; need to think on sys clock vs hw clock; latter is probably the answer). Given that `linux-embedded-hal` impls `embedded-hal` traits only, I agree it doesn't make sense.

Thoughts on getting `Rtcc` merged into `embedded-hal`? Contingent on both you and the embedded hal guys agreeing it's the way to go. I do, as I think `Rtc` is as worth being hardware agnostic as `InputPin`.

PTaylor-FluenTech, 2020-05-22:
I too wanted to use _human_ time in my project, but I didn't feel that any of the solutions (chrono, time) were the right fit. I have just released [embedded-time](https://github.com/PTaylor-FluenTech/embedded-time) to handle clocks, instants, durations more easily in embedded contexts.

Example Usage:
```rust
struct SomeClock;

impl Clock for SomeClock {
    type Rep = i64;

    fn now() -> Instant<Self> {
        // read the count of the clock
        // ...
        Instant::new(count as Self::Rep)
    }
}

impl Period for SomeClock {
    // this clock is counting at 16 MHz
    const PERIOD: Ratio<i32> = Ratio::<i32>::new_raw(1, 16_000_000);
}

// read from a Clock
let instant1 = SomeClock::now();

// ... some time passes

let instant2 = SomeClock::now();
assert!(instant1 < instant2);    // instant1 is *before* instant2

// duration is the difference between the instances
let duration: Option<Microseconds<i64>> = instant2.elapsed_since(&instant1);    

// add some duration to an instant
let future_instant = instant2 + Milliseconds(23);
// or
let future_instant = instant2 + 23.milliseconds();

assert(future_instant > instant2);
```

eldruin, 2020-05-22:
@David-OConnor I'm glad to hear that.
In the mean time I have updated my other RTC driver [`ds1307`](https://crates.io/crates/ds1307) to use the `Rtcc` trait as well.
If you want to publish an implementation of the `Rtcc` trait for linux that would be great, although I would rather not include `embedded-hal` in the name at this point.
Do you have a repository, branch or just a [gist](https://gist.github.com) where I could have a look?