rust-embedded/embedded-hal #208: blocking::spi::write_iter has a needless Default blocker?

almindor, 2020-05-09:
Why does `blocking::spi` have this [custom `Default`](https://github.com/rust-embedded/embedded-hal/blob/master/src/blocking/spi.rs#L83) trait?

It's making using things like `IntoIterator<Item = u8>` impossible with `write_iter` with errors like:

```rust
embedded_hal::blocking::spi::write_iter::Default<u8>` is not implemented for `nrf52832_hal::Spim<nrf52832_hal::nrf52832_pac::SPIM0>
```

I don't see why this default is even needed.

therealprof, 2020-05-10:
I'm struggling a bit to come up with an efficient &[u16] to &[u8] conversion that doesn't suck to test the versatility (and performance) of that approach.

almindor, 2020-05-10:
Even without conversion tho, say we want to use [Into]Iterators for greater flexibility, this Default trait is a blocker.

If there's no reason for it to exist I'd make a PR to remove + unify I2C and SPI naming (`write` vs `write_iter` currently).

almindor, 2020-05-24:
Nvm, figured it out with Disasm's help.

closed 2020-05-24 by almindor