rust-embedded/embedded-hal #215: Questions regarding 'embedded-hal'

Joe23232, 2020-05-29:
Hi I have a few questions regarding 'embedded-hal'

Afaik I am able to write Rust code and it will work on a Raspberry Pi or an Ardunio, however the machine code will be difference once when compiled. My question is am I able to write exactly the same code for an Arduino and it can easily be ported over to the Raspberry Pi without any changes to the code, for example lets say that I want an LED light bulb to turn on for 4 seconds and then turn off for a 4 seconds and the cycle repeats. Would my Rust code (not the machine code) be the exactly the same for both these two devices?

Is it production ready? Is it stable? Is there a lot of work that needs to be done?

Is it more harder to program using Rust `embedded-hall` compared to using the Arduino's code?

Am I able to do everything with `embedded-hall` compared to using Arduino's code?

TheZoq2, 2020-06-18:
Kind of a late reply, but hopefully it's still useful

For your first point, the goal is that *most* of the code should be the same, for example, your blinking LED would look something like this (pseudocode)

```rust
fn blink_led<PIN: digital::OutputPin, TIMER: timer::Timer>(pin: PIN, timer: TIMER) -> Result<()> {
    pin.set_high();
    timer.start(4.seconds());
    block!(timer.wait());
    pin.set_low();
    block!(timer.wait());
}
```

However, because microcontrollers aren't the same everywhere, especially not
when comparing the raspi to an arduino, the setup code for the pin and the
timer would differ.

So generally, you have a lot of generic code that uses the `embedded_hal` API,
and then some setup and glue code make it all work.

> Is it production ready? Is it stable? Is there a lot of work that needs to be done?

I'd say that depends. 1.0 is in alpha, and after the full release of that, I
don't think there will be any API changes. However, not everything is
implemented, so I find that you sometimes have to circumvent the `embedded_hal`
API and use the `HAL` implementors directly. So, it depends on what you want to do

> Is it more harder to program using Rust embedded-hall compared to using the Arduino's code?

Depends on what you're used to, but in general yes. Rust is a lot more strict
about its types and guarantees which does provide a lot more safety, but also
makes it harder to write the code. You also need a fairly good grasp of
generics.

> Am I able to do everything with embedded-hall compared to using Arduino's code?

Not yet, but a lot of it is there

eldruin, 2020-09-16:
Do you have any further questions or can we close this @Joe23232 ?

closed 2020-09-16 by Joe23232