rust-embedded/embedded-hal #224: Timer trait error mismatch

ryankurte, 2020-06-18:
In the timer trait, the CountDown and cancel traits share an error definition, but [`try_wait`](https://github.com/rust-embedded/embedded-hal/blob/036cfa48e42b8c7ec78dcb919b8b0d24cf6939b6/src/timer.rs#L81) and [`try_cancel`](https://github.com/rust-embedded/embedded-hal/blob/036cfa48e42b8c7ec78dcb919b8b0d24cf6939b6/src/timer.rs#L95) return different types.

Discovered in https://github.com/stm32-rs/stm32f4xx-hal/pull/172. I'm not sure of the correct change but, any change would be breaking so blocks #177.

cc. @therealprof

eldruin, 2020-06-18:
Maybe the problem in https://github.com/stm32-rs/stm32f4xx-hal/pull/172 could be solved with a different `Error` type in `CountDown`. I left a comment about it [here](https://github.com/stm32-rs/stm32f4xx-hal/pull/172#discussion_r441993522).

We could also discuss whether a common `Error` type makes sense for these two traits. Separate `Error` types are possible at the expense of fully-qualified syntax.

ryankurte, 2020-06-19:
> Maybe the problem in stm32-rs/stm32f4xx-hal#172 could be solved with a different Error type in CountDown. I left a comment about it here.

Yeah you got it, thanks!

> We could also discuss whether a common Error type makes sense for these two traits.

Given my complaint wasn't valid, and sharing error types there seems fine, I would rather not add this to the list of things to worry about at this time.

closed 2020-06-19 by ryankurte

eldruin, 2020-06-20:
Great. Glad to help.