rust-embedded/embedded-hal #233: Proposal: Not only io traits

no111u3, 2020-07-08:
For almost components `embedded-hal` has little configuration path and concentrate only on doing part - io exchange or other activity. I think `hal` means not only abstraction for `io` operations, it means abstraction of everything non chip specific operations, for example initialize and configure hardware peripheral.

First of all I provide components than needs on configuration:

- [ ] USART (UART or Serial)
- [ ] SPI
- [ ] I2C
- [ ] DMA
- [ ] ADC
- [ ] Timers
- [ ] DAC
- [ ] External Data Interface (EBI in STM32) with SDRAM and other memory settings

For second one - i try to create initialize traits structure but it is a biggest work for one developer and spends more time.

therealprof, 2020-07-08:
> I think `hal` means not only abstraction for `io` operations, it means abstraction of everything non chip specific operations, for example initialize and configure hardware peripheral.

`hal` means whatever you read into it.

> For second one - I try to create initialize traits structure but it is a biggest work for one developer and spends more time. I can to provide some types and other structures for review.

Please do. I'm really curious to see what you can come up with.

eldruin, 2022-04-13:
Initialization and configuration are explicitly defined as out-of-scope in `embedded-hal`. See [this documentation section](https://docs.rs/embedded-hal/latest/embedded_hal/#out-of-scope)
Given that I will go ahead and close this issue.
Feel free to reopen it if you have a new proposal and want to re-start the discussion.

closed 2022-04-13 by eldruin