rust-embedded/embedded-hal #302: Will the serial interface support USB CDC implementations of serial like on the Arduino Leonardo?

wt, 2021-08-01:
The normal way of interacting with an Arduino Leonardo has a few interesting features:

* The serial output goes over usb and is controlled by the same processor running your code. The arduino framework for leonardo implements a usb initialization and use the USB CDC class device for serial output.
* The autoreset is implemented by connecting to the CDC port at 1200 baud. This is also part of the USB handling in the arduino framework.

Will this hal be able to support these use cases so that platforms like the leonardo can be used ergonomically with a rust toolchain?

Thanks for your work on this project. I love it. :)

Disasm, 2021-08-01:
HAL traits are already supported in [usbd-serial](https://github.com/mvirkkunen/usbd-serial/blob/32e8e88a9c0b24791242d4ecbb9f6edd612cb8f6/src/serial_port.rs#L228-L270), but you will not be able to do blocking writes (like in Arduino) this way for "USB reasons". However, you can add a queue to this construction and write to the queue in a blocking manner while handling USB events in a different execution context.

Rahix, 2021-08-01:
The "HAL-generic interface" for USB is [usb-device](https://crates.io/crates/usb-device).  Many HALs already implement support for it.

The Arduino Leonardo you've mentioned is part of `avr-hal` which does not yet have USB support due to compiler problems.  See https://github.com/Rahix/avr-hal/issues/40.