rust-embedded/embedded-hal #364: MSRV for 1.0

Dirbaio, 2022-02-16:
Current MSRV is "1.46+", #351 bumps it to "1.54+". 

Bumping MSRV is a breaking change, so it'd be "technically" wrong to do it on a minor/patch release, this means we'll be stuck with the same MSRV for the entire lifetime of 1.0, which is not feasible (for example stable async traits will require a MSRV bump).

Perhaps we should document it as "1.54+ for now, but we may bump it in minor versions" or simply "latest stable Rust" like the [WG policy](https://github.com/rust-embedded/wg/blob/master/ops/msrv.md).

adamgreig, 2022-02-16:
> Bumping MSRV is a breaking change

I don't know how widely agreed upon this is, but it's certainly not the WG's current policy. Cargo describes it as "possibly breaking" and notes that projects might do it in a minor release: https://doc.rust-lang.org/cargo/reference/semver.html#env-new-rust

Dirbaio, 2022-02-16:
If you explicitly promise "1.0 works on 1.54+", and then in 1.0.1 you bump it to "1.60+", you've broken that promise, it's a breaking change.

If you promise "1.0 works on latest stable rust", then rust 1.60 stable comes out and you use some 1.60-only features in 1.0.1, you haven't broken that promise. It still works on "latest stable rust".

The WG MSRV policy is the 2nd, but currently this crate is the 1st.

adamgreig, 2022-02-17:
I don't think there's any intention that the promise in the README is stronger than "the current version compiles on 1.46", i.e. doesn't bind us in any future versions. The WG policy at least is clear; when and if this crate bumps MSRV is up to the HAL team so long as it stays working on latest stable. We could clarify the documentation in the README about this, but I don't think it's even mentioned in the crate documentation itself.

No argument against bumping MSRV before 1.0 for useful features like `include_str`, but I don't think there's a policy change required as part of 1.0; we're already happy to bump MSRV whenever.

commit 178f66ba (2022-02-24): Merge #368

368: Document MSRV upgrade and improve readme r=ryankurte a=eldruin

I moved some of the sections in the README out into separate documents in a `docs` folder.
I have left the section about alpha releases in there, though. I think it is important enough to grant it its placement.
I have added a document describing a policy to upgrade the MSRV, since this will eventually be important once we have released the `1.0.0` version.

TL;DR: The MSRV may be updated in major or minor releases but not on patch releases.

Closes #364
PR #368 (2022-02-24): Document MSRV upgrade and improve readme
I moved some of the sections in the README out into separate documents in a `docs` folder.
I have left the section about alpha releases in there, though. I think it is important enough to grant it its placement.
I have added a document describing a policy to upgrade the MSRV, since this will eventually be important once we have released the `1.0.0` version.

TL;DR: The MSRV may be updated in major or minor releases but not on patch releases.

Closes #364
closed 2022-02-24 by bors[bot]