rust-embedded/embedded-hal #369: Question: How to support Word: u16

milesgranger, 2022-02-26:
Hi there :wave: 

I'm just getting started with embedded Rust, and I see it is common (standard?) to mention the following [statement for FullDuplex trait](https://docs.rs/stm32l4xx-hal/latest/stm32l4xx_hal/prelude/trait._embedded_hal_spi_FullDuplex.html):

> Some SPIs can work with 8-bit and 16-bit words. You can overload this trait with different Word types to allow operation in both modes.

In [stm32l4xx-hal](https://docs.rs/stm32l4xx-hal/latest/src/stm32l4xx_hal/spi.rs.html#160) the word is `u8`. 

While attempting to write an driver (SPI) for [ISM43362](https://www.inventeksys.com/wp-content/uploads/ISM43362_M3G_L44_Functional_Spec.pdf) (pg. 18-19), the word is required to be `u16` and I'm struggling to figure out how to support both/change to u16 for transfer operations. Are there existing examples or some hints someone can direct me towards? :pray: 

Sorry if this is a terribly noob question. :smile:

burrbull, 2022-02-27:
Not all HALs have implementations of some traits.
But you could help us and make PR with implementation.
You can find examples in other repos.
See https://github.com/stm32-rs/stm32f4xx-hal/blob/master/src/serial.rs and https://github.com/stm32-rs/stm32f4xx-hal/blob/master/src/serial/hal_02.rs for example

Dirbaio, 2022-02-27:
If you need Word=u16, you'd have to add support for that in l4xx-hal.

You probably don't really need it though. With word=u8 you can do everything. if you want to send a u16, send 2 bytes instead, with `.to_be_bytes()` or `.to_le_bytes()` depending on the endianness needed.

milesgranger, 2022-02-27:
Okay, thanks. This was my impression, that within that macro, adding a parameter for the type (`u8`, `u16`) for the implementation of `FullDuplex<$type_>` but the statement gave me pause that perhaps there was another standard way of getting the same result. 

Appreciate your help.

closed 2022-02-27 by milesgranger