rust-embedded/embedded-hal #377: Tracking issue for ADC trait

eldruin, 2022-04-13:
The `ADC` traits (`adc::Channel` and `adc::OneShot`) available on the 0.2.x versions have been removed ahead of the 1.0.0 release. See: #376 
This issue servers as a tracking issue until we add them back/reject them.

The reasons were:
- The traits are quite complicated to use in generic code.
- It seems users would be better served by inherent methods.
- Also, these traits are `nb`-only.

Timer traits as of the 0.2.7 release:
- [Docs](https://docs.rs/embedded-hal/0.2.7/embedded_hal/trait.Adc.html)
- [Source](https://github.com/rust-embedded/embedded-hal/blob/2e91b430c5f3c2e06cb3f12a7812e17cafd2d3bc/src/adc.rs#L89)

Relevant issues/PRs:
- #10 
- #123 
- #101 
- #110
- Your issue here!

Please feel free to participate, highlight your current use cases, problems and provide solutions.

ryankurte, 2022-04-28:
the updated `spi` traits have me wondering if a similar approach would suit for ADCs... something like a shared `AdcDevice` that is split into separate channels implementing `AdcChannel` which drivers then take in the same manner as digital pins or any other injected type.

ADC peripherals would seem to have very similar constraints as sharing as a bus (one ADC peripheral with multiple channels that may be used in different places, and a need to be able to take ownership of the peripheral for more complex behaviors like interleaving or DMA)

Dirbaio, 2022-04-28:
Maybe having a trait for an "ADC device" is not really needed? We could just have this

```
trait AdcChannel
    fn read(&mut self) -> Result<u32, Error>;
}
```

If the hardware has a single multi-channel ADC it's up to the implementors to do the "sharing" for it. Via RefCell, Mutex, or whatever. Or it could be even a noop for MCU peripheral ADCs: hit the registers directly from `read()` and ensure the channels are `!Send` so that the user can't hit the ADC concurrently from multiple threads (irq priority levels).

> need to be able to take ownership of the peripheral for more complex behaviors like interleaving or DMA.

I don't think making traits for these is feasible, they're incredibly hardware dependent...

ryankurte, 2022-04-29:
> Maybe having a trait for an "ADC device" is not really needed? We could just have this

what i'm mostly wondering is whether this is a consistent problem with spi / i2c / adc / etc. that we can extract in a consistent way, but yeah seems to me like `AdcChannel` could be a good start.

> I don't think making traits for these is feasible, they're incredibly hardware dependent...

agreed, but the in the spi model you can still take exclusive ownership of the bus and do whatever is needed via the closure based approach.

eldruin, 2022-05-11:
A complication in the case of ADCs is that some channel combinations can be configured to work in comparison mode, where the measurements taken correspond to the difference between two (otherwise-independent) channels.
See the [ads1x1x driver](ahttps://github.com/eldruin/ads1x1x-rs) for an example.

crjeder, 2022-07-14:
> ```
> trait AdcChannel
>     fn read(&mut self) -> Result<u32, Error>;
> }
> ```

I'd strongly vote for a read() interface for ADC as this would abstract away how the ADC is connected. Could be a GPIO or via SPI.

Ben-PH, 2024-05-28:
I would like to propose making the result type generic. Perhaps the vast majority of the time it's a u32, but making it such that it can be any unsigned would make this trait more universally applicable.

darconeous, 2025-03-28:
> I would like to propose making the result type generic. Perhaps the vast majority of the time it's a u32, but making it such that it can be any unsigned would make this trait more universally applicable.

So something like this, perhaps?

```rust
trait AdcChannel {
    /// The underlying word type for ADC conversions.
    ///
    /// Typically, this is `u32` or `u16`.
    type Word: Sized + Copy;
}

trait AdcChannelOneshot : AdcChannel + ErrorType {
    /// Returns a single raw ADC reading.
    ///
    /// Calling this method will initiate an ADC conversion,
    /// block until that conversion is complete, and then return
    /// the raw result (or an error).
    ///
    /// The units are dependent on the hardware configuration,
    /// and are outside of the scope of this trait.
    fn read_oneshot(&mut self) -> Result<<Self as AdcChannel>::Word, <Self as ErrorType>::Error>;
}

trait AdcChannelOneshotMv : AdcChannel + ErrorType {
    /// Returns a single ADC reading in millivolts.
    ///
    /// Calling this method will initiate an ADC conversion,
    /// block until that conversion is complete, and then return
    /// the result (or an error).
    fn read_oneshot_mv(&mut self) -> Result<<Self as AdcChannel>::Word, <Self as ErrorType>::Error>;
}
```

And maybe some async versions:

```rust
trait AdcChannelOneshotAsync : AdcChannel + ErrorType {
    fn poll_read_oneshot(&mut self, cx: &mut Context<'_>) -> Poll<Result<<Self as AdcChannel>::Word, <Self as ErrorType>::Error>>;
}

// Include traits like `AdcChannelOneshotAsyncExt` (to provide an
// `async_read_oneshot()` method and a `AdcChannelOneshotAsyncFuture`
//  struct that would be returned by  `async_read_oneshot()`...
```

Having the ability to at least have reads be abstracted away helps make my code portable, so something like this would be very useful.

bnjmn21, 2025-11-25:
Your last point is definitely something I overlooked, and sure, `_oneshot_` can be added.
But the issue with the word being generic is that with the current trait bounds, you cannot do any arithmetic with them in driver/app code. You would need a trait bound like `std::ops::Add + std::ops::Mul + everything else` or use a solution like `num_traits::sign::Unsigned`. Additionally it should also be `Into<u32>`, and `Into<u16>`, but u32 doesn't implement `Into<u16>`. So you would need a custom `Cast<u16>` trait that does `_ as u16` casting for that. I hope you understand the problem.

darconeous, 2025-11-26:
I mean, it feels like a solvable problem.

You can easily create a trait called `AdcValue` which uses all of the required traits. You then implement `AdcValue` for all types implementing those traits. Then you only have one trait to worry about when plumbing around. Seems easy enough.

darconeous, 2025-11-26:
Regarding casting, that seems like an additional constraint that could/should be added to any generics that need to do that.

Realistically, I think you'd want to use `TryInto<>` instead of `Into<>` anyway. And `u32` does implement `TryInto<u16>`.

Dirbaio, 2025-11-26:
> @Ben-PH: ZCA is a core tenant of rust. Unless it's explicitly stated as a non goal to provide ZCA, then the goal of ZCA is implied.

You're telling me, a maintainer of this project, what the goal of this project is?

The goal of `embedded-hal` is to produce traits that allow any driver to work with any HAL. If these traits can be "zero cost" then great, but "zero cost" is secondary. It comes after the "allow any driver to work with any HAL" goal.

> @darconeous; Should I just whip up a pull request for my original version?

No, it has an unconstrained `Word` associated type. This makes it very unwieldy or even impossible to write a driver that works with all HALs, if one HAL implements Word=u32 and another implements it with Word=u16. See [here](https://github.com/rust-embedded/embedded-hal/blob/master/docs/migrating-from-0.2-to-1.0.md#unconstrained-associated-types).

Ben-PH, 2025-12-03:
To me, the idea that asking for an embedded framework doesn't just assume everythibg is in u32 is "a hill to die on" is straight up bizarre.

bnjmn21, 2025-12-04:
@darconeous Can you please explain again what traits you would add to `AdcValue`? I'm not sure I understood you correctly. Because if you just do `Into<u32> + TryInto<u16>`, I don't think it's worth it, because `Into` / `TryInto` aren't zero-cost, all they do is make the cast explicit. `num_traits::Unsigned` is the only way to make it zero-cost, because then you could do arithmetic directly on the generic type, without any casting. (But I don't think that's a good idea, because it adds even more complexity).

Beanow, 2025-12-04:
Something to add to the topic of runtime performance costs.

I'm sure some of you already had this in mind, but this trait may also want to consider adding however many constants as necessary describing the ADC, to make `const` math more ergonomic, or easy for a general crate to work with.

Especially to avoid floating point math at runtime. But perhaps also integer divisions.

Imagining some pseudo code macro here.
```rs
if reading >= volt!(1.15, ConcreteAdc) && reading <= volt!(1.55, ConcreteAdc) {
  // Between 1.15 and 1.55 volt logic
}

// Expanded macro does const math, which gets optimized to a specific integer value
if reading >= 201 && reading <= 352 {
  // Between 1.15 and 1.55 volt logic
}

```

darconeous, 2025-12-05:
@bnjmn21: My idea was perhaps a bit half baked. Rust is great for hinting that something will be easy when instead it ends up being bonkers. But  let's roll up our sleeves and see where this rabbit hole leads... Maybe it ends up working the way I wanted it to.

The idea was that you would have a `AdcChannel` trait that would look like this:

```rust
pub trait AdcChannel {
    /// The underlying word type for ADC conversions.
    ///
    /// Typically, this is `u32` or `u16`.
    type Word: AdcWord;
}
```

...And then have a trait called `AdcWord` which would have all of the traits implemented for supporting operations. These are all the typical traits that `u32` and `u16` (and hell maybe even `i16`) have in common. Yes, **there are A LOT**, but macros can help, and once it is written you don't have to deal with it anymore (except *maybe* adding to it on occasion).

I imagine it would look something kinda like this (if we don't end up defining it with a macro, which we likely should):

```rust
use core::ops::Add;
use core::ops::AddAssign;
// ... snip ...
use core::fmt::UpperHex;
use core::fmt::LowerHex;

pub trait AdcWord: Sized
    + Copy
    + TryInto<u32>
    + TryInto<u16>
    + Add
    + for<'a> Add<&'a Self>
    + AddAssign
    + for<'a> AddAssign<&'a Self>
    // ... snip ...
    + UpperHex
    + LowerHex
{}
```

Then you'd need to make sure that `AdcWord` gets implemented automatically for any trait implementing *all of those traits*. So, something like this:

```rust
impl <T> AdcWord for T where T: Sized
    + Copy
    + TryInto<u32>
    + TryInto<u16>
    + Add
    + for<'a> Add<&'a Self>
    + AddAssign
    + for<'a> AddAssign<&'a Self>
    // ... snip ...
    + UpperHex
    + LowerHex
{}
```

This should, in theory, give you something that very nearly supports everything that you might want to do from a generic, and if you want to do something that happens to fall outside of this list you just bite the bullet and do a `.try_into<u16>()` or `.try_into<u32>()`...

What's nice about this approach is that the long list of traits is an implementation detail that the user of the crate doesn't have to worry about. Generics that use `AdcChannel` types don't need to worry about that big long list, they would just bind the types using the `AdcWord` trait. Simple. Ergonomic.

Now, the list of traits in `AdcWord` is long, but it is maintainable with macros, and once it is written you are largely done. It allows generics to directly operate on ADC values in a zero-cost way most of the time. In the weird case where that isn't possible (and there are a few), the generic can call `.try_into::<u32>()` at a cost that is no worse than if we had ended up forcing everyone to use `u32`.

I've thrown this into Rust Playround and it seems to compile at least. It will take some effort to put together the full list of traits, but I personally feel like it is worth it. Once it is done it is done.

As an added bonus you don't need to rely on the `num_traits` crate. You could add it as an optional addition if you wanted to use those traits in generics too, but eh.

So that wasn't too bad. Phew.

I really think this idea is worth considering. It sure seems like all upside (possibility of zero-cost generics) and no downside (when that isn't possible, it's no worse than if we had just forced everyone to use `u32`).