rust-embedded/embedded-hal #387: It's not clear when `can::ErrorKind::Acknowledge` should be returned

jonas-schievink, 2022-05-25:
Its docs say "An ACK error shall be detected by a transmitter whenever it does not monitor a dominant bit during the ACK slot.", but `Can::transmit` only enqueues a frame for transmission, so it's impossible to detect this condition for the frame passed there. `Can::transmit` *could* check if any previous frame caused an ACK error and return an error, but then its return value indicates whether a *different* frame was transmitted successfully.

eldruin, 2022-06-06:
cc: @timokroeger

timokroeger, 2022-06-06:
I fully agree with jonas here.
The error kinds were chosen to map to the error conditions defined by the CAN spec (same wording in the doc strings as the description in the spec).
For our `transmit()` interface returning an ACK error does not make sense as those should be handled internally by the controller.

eldruin, 2022-07-05:
We discussed in the WG meeting that we could keep the variant but improve its documentation.
Any change suggestions?