rust-embedded/embedded-hal #395: Some questions regarding this project.

Raj2032, 2022-07-31:
Hi I have some questions regarding this project.

Is this a "write once run everywhere" project? That means that if I write code for the Arduino, the exact same code would work on a Raspberry Pi, for example if I wanted to make an LED blink I can use the exact same code, the only difference is that the binary output after compiling would be very different between the two devices?

If I had to different Arduino models, would the code be exactly the same?

Fomys, 2022-07-31:
The code will not be exactly the same, you have different device initialization on each devices.
This crate provide common traits to write generic code. It's mainly useful when writing drivers.
For example, if you want to make a driver "Led", you need one pin, this pin will be "PA1" on stm32, "D13" for arduino, ..., but your driver can be generic on "OutputPin".

For example this driver is compatible with all crates that provide the traits from embedded-hal :
```rust
struct Led<P: OutputPin> {
    pin: P
}

impl<P: OutputPin> Led<P> {
    fn new(pin: P) -> Self {
        Self {
            pin,
        }
    }

    fn on(&mut self) {
        self.set_high();
    }

    fn off(&mut self) {
        self.set_low();
    }
}```

Raj2032, 2022-08-01:
struct Led<P: OutputPin> {
    pin: P
}

impl<P: OutputPin> Led<P> {
    fn new(pin: P) -> Self {
        Self {
            pin,
        }
    }

    fn on(&mut self) {
        self.set_high();
    }

    fn off(&mut self) {
        self.set_low();
    }
}```

So if I wrote this for both the Arduino and Raspberry Pi, this code can remain exactly the same to control an LED, am I understanding you correctly?

Fomys, 2022-08-01:
This part of code yes, but the way you have to initialize the GPIO no

Raj2032, 2022-08-19:
@Fomys Hey man sorry for the late reply.

So in regards to a blinking led program for both a raspbierry pi pico and arduino.

Here is an example of [avr/arduino](https://github.com/Rahix/avr-hal/blob/main/examples/arduino-uno/src/bin/uno-blink.rs#L18)
```rust
    loop {
        led.toggle();
        arduino_hal::delay_ms(100);
        led.toggle();
        arduino_hal::delay_ms(100);
        led.toggle();
        arduino_hal::delay_ms(100);
        led.toggle();
        arduino_hal::delay_ms(800);
    }
```

And an example for the [raspberry pi pico](https://github.com/rp-rs/rp-hal/blob/main/rp2040-hal/examples/blinky.rs#L85)

```rust
    loop {
        led_pin.set_high().unwrap();
        // TODO: Replace with proper 1s delays once we have clocks working
        delay.delay_ms(500);
        led_pin.set_low().unwrap();
        delay.delay_ms(500);
    }
```

as you can see they are different. Is there a way they can be written generically?

Fomys, 2022-08-19:
Your snippets don't do exactly the same thing.

Both of the crate you linked already implements traits from embedded-hal, but not the same version, so it is not possible to make generic code. You can try to bump the version in avr-hal to make have the same version.

If the both crates uses the same embedded-hal version you can create this kind of generic function:

```rust
fn blink_loop<P: OutputPin>(led: P) {
    loop {
        led.setHigh().unwrap();
        for _ in 0..1000000 {}
        led.setLow().unwrap();
        for _ in 0..1000000 {}
    }
}
```

And in your executable:

```rust
fn main() {
    // ... specific code for your target board
    let led = /* specific code for your target board */ .into_push_pull_output();
    
    blink_loop(led);
}
```

The first method is generic, so you can use it with anything that implement the traits from embedded-hal crate.

Fomys, 2022-08-24:
> Right, so like what does toggle() do compared to set_high()/set_low() don't they both turn on and off the pin?
They both blink the leds, but not at the same speed. toggle() means "change the output state to the oposite one"
> Another question, does embedded-hal support as many various gpio sensors/devices (such as LED, ultrasonic sound distance sensor etc) as the Arduino IDE code/library does?
I think not, but you can search on crates.io if someone already write a driver for your device.

> You may be able to find even more HAL implementation crates and driver crates by searching for the keywords [embedded-hal-impl](https://crates.io/keywords/embedded-hal-impl), [embedded-hal-driver](https://crates.io/keywords/embedded-hal-driver) and [embedded-hal](https://crates.io/keywords/embedded-hal) on crates.io.

Raj2032, 2022-08-25:
Right i see thanks mate :)

eldruin, 2022-08-31:
The original question seems solved. Thanks @Fomys! Closing.

closed 2022-08-31 by eldruin