rust-embedded/embedded-hal #417: Why is there no spi::blocking::Read trait

peckpeck, 2022-10-07:
There is a blocking write trait, but it looks to me that the read is missing

Dirbaio, 2022-10-07:
The upcoming embedded-hal 1.0 does have SPI read. 

In 0.2 you can use `transfer` with a zerofilled buffer.

peckpeck, 2022-10-08:
Thank you
as for transfer, it seems to only work if there are as many bytes sent as received (or it is not well documented enough)

adamgreig, 2022-10-08:
That's right (at least in the current 0.2 version of embedded-hal). It's because SPI must always send as many bytes as it receives. You can just put dummy bytes to make sure you read as much as you want (and remember the first bytes "received" will be during the initial transmission, so you probably want to discard them).

peckpeck, 2022-10-09:
i'm not sure the spi peripheral i'm using respects this kind of rule
but as long as what i need is in the next version, all is good for me

closed 2022-10-12 by peckpeck