rust-embedded/embedded-hal #427: Implement Into<bool> for PinState

sourcebox, 2022-11-21:
`From<bool>` is already implemented for PinState in `digital.rs`, which makes sense for output pins. For input pins, it would make sense to implement the `Into<bool>` trait as well, so a function could take an argument that is either a `PinState` or a `bool`:

```rust
impl Into<bool> for PinState {
    fn into(self) -> bool {
        match self {
            PinState::Low => false,
            PinState::High => true,
        }
    }
}

fn print_state<T: Into<bool>>(state: T) {
    println!("{:?}", state.into());
}

fn main() {
    print_state(false);
    print_state(true);
    print_state(PinState::Low);
    print_state(PinState::High);
}
```

Dirbaio, 2022-11-21:
It should be `impl From<PinState> for bool`, as you should never impl `Into` manually since it's derived from `From`. Otherwise SGTM! PR welcome :)

sourcebox, 2022-11-21:
> It should be `impl From<PinState> for bool`, as you should never impl `Into` manually since it's derived from `From`. Otherwise SGTM! PR welcome :)

Yes, you're right. So take this:

```rust
impl From<PinState> for bool {
    fn from(value: PinState) -> bool {
        match value {
            PinState::Low => false,
            PinState::High => true,
        }
    }
}
```

commit c7ac0df0 (2022-12-20): Merge #429

429: Implement From<PinState> for bool r=Dirbaio a=sourcebox

Closes #427
PR #429 (2022-12-20): Implement From<PinState> for bool
Closes #427
closed 2022-12-20 by bors[bot]