rust-embedded/embedded-hal #454: Broken documentation for GPIO23

VorpalBlade, 2023-04-26:
According to rust-analyzer I'm getting a Gpio23 object (which makes sense, I'm trying to hook up a LED to that GPIO for a test). When I try to find documentation for it however at https://esp-rs.github.io/esp-idf-hal/esp_idf_hal/gpio/index.html it only goes up to Gpio21!?

I presume this is due to this docs being built for a different module, but on docs.rs there is a dropdown to switch between build targets. It would be really useful to have that here.

For context: I'm using and targeting a traditional Xtensa ESP32 module (ESP32-WROVER-IE specifically on a ESP32-DevKitC, this was the only dev board with I2S support and traditional non-LE Bluetooth I could find in stock here in Sweden).

Dirbaio, 2023-04-26:
You should file this issue at https://github.com/esp-rs/esp-idf-hal/

closed 2023-04-26 by Dirbaio

VorpalBlade, 2023-04-26:
Sorry, too many tabs open while trying to figure out the mess that is embedded rust.