rust-embedded/embedded-hal #479: embedded-hal-bus RefCellDevice should use Cell<Option<&mut I2c>> instead

Kry-Vosa, 2023-08-05:
The whole idea is in the title: Since we do not need multiple immutable borrows at the same time, we can use just `Cell<Option<&mut I2c>>`.

"Borrowing" the reference is just:
```rust
let borrowed = cell.replace(None).unwrap();
cell.replace(Some(borrowed));
```
(Or something more fancy if you wish)

As for why should we do that, it will save a whole `isize` of memory, which might be actually meaningful on devices like Arduino, where memory is scarce and likelihood of multiple I2C drivers high.

Kry-Vosa, 2023-08-05:
Even better might be another impl with [qcell::LCell](https://docs.rs/qcell/latest/qcell/struct.LCellOwner.html), mainly to point users to the existing zero-cost abstraction if they really need it.

Dirbaio, 2023-08-05:
- This is a bit less ergonomic because you now need 2 variables: the i2c, and then the `Cell<Option<I2c>>` pointing to the 1st, and you have to ensure the 1st outlives the 2nd.
- It's not always a win: you remove the `isize` RefCell borrow flag, but you add a pointer that you didn't have before. It's a tie if the i2c size is a multiple of `usize` size, it's a slight win otherwise due to alignment but in practice it's unlikely to make a difference (allocations are likely to get rounded up to usize if the thing coming after needs usize alignment). See https://play.rust-lang.org/?version=stable&mode=debug&edition=2021&gist=ed5a2507ea3503f2526ea2553333c503

So I don't think it's worth it, I'd prefer to keep the simpler, more idiomatic RefCell.

Also, note that nothing prevents external crates from implementing `SpiDevice` with `Cell<Option>` or `LCell` or anything else. It doesn't *have* to be in `embedded-hal-bus`, it'll still interoperate with the ecosystem fine thanks to the `SpiDevice` trait.

Kry-Vosa, 2023-08-05:
You are indeed correct and I'm dumb. Sorry for the wasted time.
However I believe the point with `LCell` still stands, it would be *nice* if `embedded-hal-bus` provided zero-cost way of sharing buses instead of forcing developers to look elsewhere.

closed 2023-08-05 by Kry-Vosa