rust-embedded/embedded-hal #480: New embedded-io release

eldruin, 2023-08-06:
There has been some refactoring/renaming in the `embedded-io` traits during the integration into this repository.
It would be good to release a new version including these changes.
Is there anything left to do before its release?

Dirbaio, 2023-08-06:
I think it should be good to go. Perhaps we can release them as `1.0.0-rc1`?

eldruin, 2023-08-06:
Good question. Since there have been so many changes, how about releasing 0.5 first and after people have migrated then promote it to rc1?

Dirbaio, 2023-08-06:
isn't that the point of rc's? if we think the crate is ready for 1.0, we signal that by relasing 1.0-rc1 to request people to try it out and give feedback.

Otherwise we're forcing people to do 2 breaking changes: 0.5 -> 1.0-rc1, and then 1.0-rc1 -> 1.0

eldruin, 2023-08-06:
I was trying to avoid the situation where there are breaking changes between an rc1 and a _potential_ rc2 release.
However, I have not been involved in `embedded-io`'s history so I would accept your assessment if you are sure it is ready for rc1 and no breaking changes are expected.

Dirbaio, 2023-08-06:
okay, let's do 0.5 then. I can't think of any issues, but it's true that ReadReady/WriteReady are new, and also that pointing embedded_hal::Serial users to it will bring more eyes with perhaps new use cases.

I was trying to update Embassy to 0.5, and one thing we've lost since 0.4 is `defmt` support. I'll send a PR.

Dirbaio, 2023-08-06:
#481