rust-embedded/embedded-hal #504: Why do the SPI/I2C transaction slices have to be mutable?

Dominaezzz, 2023-09-09:
In the [SPI transaction API](https://github.com/rust-embedded/embedded-hal/blob/030a2324e1a7ea833372679bc1f32cdda9ba6494/embedded-hal/src/spi.rs#L353) and the [I2C transaction API](https://github.com/rust-embedded/embedded-hal/blob/master/embedded-hal/src/i2c.rs#L384) a mutable slice of operations is taken but I don't see why this is needed.

See [Rust Playground](https://play.rust-lang.org/?version=stable&mode=debug&edition=2021&gist=586997a15202a258eb3c2022bd234490) example showing `mut` isn't necessary.

Rahix, 2023-09-09:
Your playground example isn't complete as you are missing the immutable reference.  Try putting the `for` loop into a function which takes `&[...]` and you'll notice it does not work anymore: https://play.rust-lang.org/?version=stable&mode=debug&edition=2021&gist=510670b534ad0a4fb665cd5aa2831362

The reason is that in your example the for loop operates on the owned array of operations (`[Operation; 2]`) while the function takes a reference to a slice (`&[Operation]`).  In the latter case, even though there are mutable references inside the slice, the outer `&` prevents mutation of the data.

Dominaezzz, 2023-09-09:
Ahhhh I see, thanks for the explanation.

closed 2023-09-09 by Dominaezzz