rust-embedded/embedded-hal #509: non blocking read

JonasFocke01, 2023-09-27:
Hello :),

i have a i2c sensor that is build in a way, that it can reset at any time. Sadly, this stops my rust code as well, because it could be in the middle of a transaction to that sensor, and waits for data that is never coming. I dug around in the code, but i have not found a way to implement a non_blocking_read() function alongside the normal read. Do you know what it would take to do that? I'm willing to help, but im not smart enough it seems :sweat_smile:

Dirbaio, 2023-09-27:
Some HALs (for example Embassy's) have support for setting an i2c timeout, so the blocking i2c trait will return with an error instead of waiting forever. This is something the HAL does, that alters the behavior of the existing i2c blocking trait (there's no "i2c with timeout" trait).

You can also use the `embedded-hal-async` i2c trait. This allows doing something like `with_timeout(Duration::from_secs(1), i2c.read(...)).await`, which will return early with an error if the timeout is reached, without HAL collaboration.

JonasFocke01, 2023-09-28:
Thanks for your response! I will try the methods you mentioned.  
The question that remains in my head is now, why is the watchdog not triggering? Or in other words, what would be needed for the watchdog to trigger on this?

Dirbaio, 2023-09-28:
Watchdogs typically aren't enabled by default, you need to enable it and then feed/pet it periodically.

JonasFocke01, 2023-09-29:
That worked really well, thank you :grinning: :heart:

closed 2023-09-29 by JonasFocke01