rust-embedded/embedded-hal #571: embedded-hal-bus shared i2c usage

pdgilbert, 2024-01-21:
Using `embedded-hal_1.0.0` I have been trying to replace `shared_bus` in several examples. @rursprung pointed out the this can be done with `embedded-hal-bus`. I am finding it is not a simple replacement. 

I have been using `RefCellDevice`. The code below compiles with `stm32f4xx_hal` and with `stm32h7xx-hal` branch `eh-v1.0` as updated by @richardeoin. It uses crates
- @eldruin  `ads1x1x`   (git = "https://github.com/eldruin/ads1x1x-rs" )
- @tdittr   `ina219 `   (git = "https://github.com/tdittr/ina219", branch = "add-configuration-register")
- @bugadani `ssd1306`  (git = "https://github.com/bugadani/ssd1306", branch = "ehal1" )

all of which use `embedded-hal_1.0.0`.

I have not tried to run on hardware. I am fairly sure it will not work, based on my confusion and the implication of the GIGO principle. (And also the fact that it does not pass the bus address to `ssd1306`.)

My coding effort has been guided by
- embedded_hal_bus 0.1.0 documentation https://docs.rs/embedded-hal-bus/0.1.0/embedded_hal_bus/i2c/struct.RefCellDevice.html
- and https://docs.rs/embedded-hal-bus/0.1.0/embedded_hal_bus/i2c/index.html
- and comments in https://docs.rs/embedded-hal-bus/0.1.0/src/embedded_hal_bus/i2c/refcell.rs.html#70-76
- Migrating from embedded-hal 0.2.x to 1.0.0 https://github.com/rust-embedded/embedded-hal/blob/master/docs/migrating-from-0.2-to-1.0.md
- recent examples by @reitermarkus and @eldruin in https://github.com/eldruin/ads1x1x-rs//examples/.

but I am still confused. That is probably because I am missing some basic point everyone thinks is so obvious that there is no need to mention it.

I would very much appreciate any explanations/comments/advice. My main confusion is over what parts of the implementation should be done in device crates and what needs to be done in user code. I have a poor understand of what trait implementation coherence and the orphan rule imply for where things must be done in  the hals, device crates, and user code.

The code compiles with a section that does more of the implementation than I think is needed in user code. Comments in the code indicate that section as well as something closer to what I think the user should need. Some other questions are also indicated in comments.

<details>
<summary>Click to expand code </summary>

```
#![deny(unsafe_code)]
#![no_std]
#![no_main]
#![feature(type_alias_impl_trait)]

#[cfg(debug_assertions)]
use panic_semihosting as _;

#[cfg(not(debug_assertions))]
use panic_halt as _;

use cortex_m_rt::entry;
use embedded_hal::delay::DelayNs;

/////////////////////   ads
use ads1x1x::{Ads1x1x, channel, ChannelSelection, DynamicOneShot, FullScaleRange, SlaveAddr};

/////////////////////   ina
use ina219::{address::{Address, Pin}, measurements::BusVoltage, SyncIna219};

/////////////////////   ssd
use ssd1306::{mode::BufferedGraphicsMode, prelude::*, I2CDisplayInterface, Ssd1306};
const DISPLAYSIZE:ssd1306::prelude::DisplaySize128x32 = DisplaySize128x32;
const VPIX:i32 = 12; // vertical pixels for a line, including space

use core::fmt::Write;
use embedded_graphics::{
    mono_font::{ascii::FONT_6X10 as FONT, MonoTextStyle, MonoTextStyleBuilder},
    pixelcolor::BinaryColor,
    prelude::*,
    text::{Baseline, Text},
};

/////////////////////   hals

#[cfg(feature = "stm32f4xx")]
pub use stm32f4xx_hal::{
   pac::{Peripherals, I2C1},
   rcc::{Clocks, RccExt},
   pac::{TIM2, TIM5},
   timer::Delay,
   timer::TimerExt,
[... 356 lines cut ...]
    Finished dev [unoptimized + debuginfo] target(s) in 19.69s

```
</details>

<details>
<summary>Click to expand compile results with stm32h7xx_hal</summary>

```
$ cargo build --no-default-features --target thumbv7em-none-eabihf --features stm32h742,stm32h7xx --example ads_ina_ssd
   Compiling nb v0.1.3
   Compiling paste v1.0.14
   Compiling stm32h7 v0.15.1
   Compiling cast v0.3.0
   Compiling embedded-io v0.6.1
   Compiling embedded-hal v0.2.7
   Compiling cortex-m v0.7.7
   Compiling cortex-m-semihosting v0.5.0
   Compiling panic-semihosting v0.6.0
   Compiling stm32h7xx-hal v0.15.1 (https://github.com/stm32-rs/stm32h7xx-hal?branch=eh-v1.0#b05811ce)
   Compiling embedded-hal-bus_example v0.0.1 (/ads_ina_ssd)
    Finished dev [unoptimized + debuginfo] target(s) in 22.18s

```
</details>

pdgilbert, 2024-01-21:
For completeness, here are error messages when compiling with the section I think should work:

<details>
<summary>Click to expand compile errors with section I think should work (stm32f4xx_hal)(</summary>

```
 $ cargo build --no-default-features --target thumbv7em-none-eabihf --features stm32f401,stm32f4xx --example ads_ina_ssd
   Compiling embedded-hal-bus_example v0.0.1 (/ads_ina_ssd)
warning: unused imports: `ErrorKind`, `ErrorType`, `SevenBitAddress`
   --> examples/ads_ina_ssd.rs:109:30
    |
109 | use embedded_hal::i2c::{I2c, SevenBitAddress, ErrorKind, ErrorType}; //...
    |                              ^^^^^^^^^^^^^^^  ^^^^^^^^^  ^^^^^^^^^
    |
    = note: `#[warn(unused_imports)]` on by default

warning: unused import: `embedded_hal_bus::i2c`
   --> examples/ads_ina_ssd.rs:110:5
    |
110 | use embedded_hal_bus::i2c;
    |     ^^^^^^^^^^^^^^^^^^^^^

error[E0277]: the trait bound `embedded_hal_bus::i2c::RefCellDevice<'_, impl embedded_hal::i2c::I2c>: embedded_hal::i2c::I2c` is not satisfied
   --> examples/ads_ina_ssd.rs:244:42
    |
244 | ...et mut adc_a = Ads1x1x::new_ads1015(adc_a_rcd, SlaveAddr::Alternativ...
    |                   -------------------- ^^^^^^^^^ the trait `embedded_hal::i2c::I2c` is not implemented for `embedded_hal_bus::i2c::RefCellDevice<'_, impl embedded_hal::i2c::I2c>`
    |                   |
    |                   required by a bound introduced by this call
    |
help: trait impl with same name found
   --> .cargo/git/checkouts/embedded-hal-2c6e880c6b254bb0/04b4f18/embedded-hal-bus/src/i2c/refcell.rs:85:1
    |
85  | / impl<'a, T> I2c for RefCellDevice<'a, T>
86  | | where
87  | |     T: I2c,
    | |___________^
    = note: perhaps two different versions of crate `embedded_hal` are being used?
    = help: the following other types implement trait `embedded_hal::i2c::I2c<A>`:
              <I2cStub as embedded_hal::i2c::I2c>
              <stm32f4xx_hal::i2c::I2c<I2C> as embedded_hal::i2c::I2c>
              <&mut T as embedded_hal::i2c::I2c<A>>
note: required by a bound in `ads1x1x::construction::<impl Ads1x1x<I2C, Ads1015, Resolution12Bit, ads1x1x::mode::OneShot>>::new_ads1015`
   --> .cargo/git/checkouts/ads1x1x-rs-bbbb6dc8bfa9c063/5cd3613/src/construction.rs:39:1
    |
39  | impl_new_destroy!(Ads1015, new_ads1015, destroy_ads1015, ic::Resolution12Bi...
    | ^^^^^^^^^^^^^^^^^^^^^^^^^^^-----------^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    | |                          |
    | |                          required by a bound in this associated function
    | required by this bound in `ads1x1x::construction::<impl Ads1x1x<I2C, Ads1015, Resolution12Bit, OneShot>>::new_ads1015`
    = note: this error originates in the macro `impl_new_destroy` (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0277]: the trait bound `embedded_hal_bus::i2c::RefCellDevice<'_, impl embedded_hal::i2c::I2c>: embedded_hal::i2c::I2c` is not satisfied
   --> examples/ads_ina_ssd.rs:245:42
    |
245 | ...et mut adc_b = Ads1x1x::new_ads1015(adc_b_rcd, SlaveAddr::Alternativ...
    |                   -------------------- ^^^^^^^^^ the trait `embedded_hal::i2c::I2c` is not implemented for `embedded_hal_bus::i2c::RefCellDevice<'_, impl embedded_hal::i2c::I2c>`
    |                   |
    |                   required by a bound introduced by this call
    |
help: trait impl with same name found
   --> .cargo/git/checkouts/embedded-hal-2c6e880c6b254bb0/04b4f18/embedded-hal-bus/src/i2c/refcell.rs:85:1
    |
85  | / impl<'a, T> I2c for RefCellDevice<'a, T>
86  | | where
87  | |     T: I2c,
    | |___________^
    = note: perhaps two different versions of crate `embedded_hal` are being used?
    = help: the following other types implement trait `embedded_hal::i2c::I2c<A>`:
              <I2cStub as embedded_hal::i2c::I2c>
[... 747 lines cut ...]
help: trait impl with same name found
   --> .cargo/git/checkouts/embedded-hal-2c6e880c6b254bb0/04b4f18/embedded-hal-bus/src/i2c/refcell.rs:85:1
    |
85  | / impl<'a, T> I2c for RefCellDevice<'a, T>
86  | | where
87  | |     T: I2c,
    | |___________^
    = note: perhaps two different versions of crate `embedded_hal` are being used?
    = help: the following other types implement trait `embedded_hal::i2c::I2c<A>`:
              <I2cStub as embedded_hal::i2c::I2c>
              <stm32h7xx_hal::i2c::I2c<stm32h7xx_hal::stm32::I2C1> as embedded_hal::i2c::I2c>
              <stm32h7xx_hal::i2c::I2c<stm32h7xx_hal::stm32::I2C2> as embedded_hal::i2c::I2c>
              <stm32h7xx_hal::i2c::I2c<stm32h7xx_hal::stm32::I2C3> as embedded_hal::i2c::I2c>
              <stm32h7xx_hal::i2c::I2c<stm32h7xx_hal::stm32::I2C4> as embedded_hal::i2c::I2c>
              <&mut T as embedded_hal::i2c::I2c<A>>
    = note: required for `Ads1x1x<RefCellDevice<'_, ...>, ..., ..., ...>` to implement `DynamicOneShot`
    = note: the full type name has been written to '/ads_ina_ssd/target/thumbv7em-none-eabihf/debug/examples/ads_ina_ssd-5d0addd19a929217.long-type-16730457139217183598.txt'

Some errors have detailed explanations: E0277, E0599.
For more information about an error, try `rustc --explain E0277`.
warning: `embedded-hal-bus_example` (example "ads_ina_ssd") generated 2 warnings
error: could not compile `embedded-hal-bus_example` (example "ads_ina_ssd") due to 15 previous errors; 2 warnings emitted

```
</details>

bugadani, 2024-01-22:
The problem is this line in your `Cargo.toml`:

`embedded-hal-bus = { git = "https://github.com/rust-embedded/embedded-hal/" }`

You can't mix and match dependencies from `crates.io` and `git`. This import causes a second `embedded-hal` to be imported, which defines a (from the compiler's viewpoint) different `I2c` trait than the STM32 crates implement, or the device crates expect.

The fix is as simple as replacing it with a version from `crates.io`:

`embedded-hal-bus = "0.1"`

pdgilbert, 2024-01-22:
Thank you @bugadani . When I use
```
embedded-hal = "1.0"
embedded-hal-bus = "0.1"
```
the code compiles without error using the section I thought should work. I am still not passing  the bus address to `ssd1306`. If I change from
```
    let interface = I2CDisplayInterface::new(ssd_rcd); 
```
to

```
    let interface = I2CDisplayInterface::new(ssd_rcd, 0x3C); 
```
I get 
```
error[E0061]: this function takes 1 argument but 2 arguments were supplied
   --> examples/ads_ina_ssd.rs:256:21
    |
256 | ...nterface = I2CDisplayInterface::new(ssd_rcd, 0x3C);  // check address 
    |               ^^^^^^^^^^^^^^^^^^^^^^^^        ------
    |                                               | |
    |                                               | unexpected argument of type `{integer}`
    |                                               help: remove the extra argument
    |
note: associated function defined here
   --> /home/paul/.cargo/git/checkouts/ssd1306-407b1a92e31533f1/0bae3a6/src/i2c_interface.rs:13:12
    |
13  |     pub fn new<I>(i2c: I) -> I2CInterface<I>
    |            ^^^
```

Also, if I specify both hals from git:
```
embedded-hal = { git = "https://github.com/rust-embedded/embedded-hal/" }
embedded-hal-bus = { git = "https://github.com/rust-embedded/embedded-hal/" }

```
I get errors as before. I guess this is probably due to the git versions of `ads1x1x`, `ina219`, and `ssd1306` all using the release version of `embedded-hal`.  It seems mixing and matching traits is more sensitive to versions than mixing and matching types.

Does this mean that all dependencies in a project must use the same version of `embedded-hal`?

And, does it mean that update rules will become a problem? For example `ads1x1x` specifies `embedded-hal = "1"`  while `ssd1306` specifies `embedded-hal = "1.0.0"`. Eventually some device crates will specify '1.0.1'. If I understand `cargo update`, which is questionable, this can result in a mix of versions.

If the last is true, perhaps there should be a preferred way to specify the dependency to help avoid mixing versions?

pdgilbert, 2024-01-23:
Thanks @bugadani . The fog is beginning to clear, there were a few things I did not understand correctly.