rust-embedded/embedded-hal #572: embedded-hal-bus does not impliment embedded-hal-async traits, despite claiming to

Sycrosity, 2024-01-22:
Despite it being stated that the `async` feature "enables embedded-hal-async support", there is no code implementation of this. Async shared busses would be very useful, especially for crates which only support the embedded-hal-async I2C traits, to be able to safely share i2c busses over.

Dirbaio, 2024-01-22:
There's [one for ExclusiveDevice](https://github.com/rust-embedded/embedded-hal/blob/04b4f1897a126d1f3aebf679489b60d8e994c839/embedded-hal-bus/src/spi/exclusive.rs#L94).

The problem with async shared buses is there's no "standard" async mutex, so it's left for crates from the ecosystem for now. For example there's [embassy-embedded-hal](https://docs.embassy.dev/embassy-embedded-hal/git/default/shared_bus/index.html) containing impls using [embassy-sync's async Mutex](https://docs.embassy.dev/embassy-sync/git/default/mutex/struct.Mutex.html).

Sycrosity, 2024-01-22:
> There's [one for ExclusiveDevice](https://github.com/rust-embedded/embedded-hal/blob/04b4f1897a126d1f3aebf679489b60d8e994c839/embedded-hal-bus/src/spi/exclusive.rs#L94).
That slipped past me, there indeed is - however there isn't any for I2C.

> The problem with async shared buses is there's no "standard" async mutex, so it's left for crates from the ecosystem for now. For example there's [...]

That seemed to be the issue when and around searching this issue - do you know of an example of a project using embedded asynchronous shared mutex's? Thank you for the links, I will investigate further into these.

liarokapisv, 2024-04-10:
Does the above have implications for the async implementation of the RefCellDevice ?

Dirbaio, 2024-04-10:
RefCell is not async, so RefCellDevice can't implement the async traits. if it did you'd get BorrowMutErrors if you tried to

liarokapisv, 2024-04-10:
RefCellDevice is useful even if only used by a single task. Eg you could implement a fair round-robin scheduler for the different devices in a single task.

elpiel, 2025-08-07:
I'd like to see the async I2c std mutex implementation, as I needed to use it recently and realised that it's missing:
 https://github.com/rust-embedded/embedded-hal/blob/master/embedded-hal-bus/src/i2c/mutex.rs#L29

Dirbaio, 2025-08-07:
How'd that look? std has no async mutex.