rust-embedded/embedded-hal #578: CAN FD support

tristanguerragomez, 2024-02-28:
[CAN FD](https://en.m.wikipedia.org/wiki/CAN_FD) is rapidly becoming that de facto as it can provide much more bandwidth and is backwards compatible with CAN 2.0.

Are there any plans to update the API to support this scenario?

reneherrero, 2024-02-28:
The fact that they moved the CAN code to a separate crate which is **not** at v1.0 might be indicative that changes are welcome?

Does anyone have any insight on this?

liamkinne, 2024-11-14:
I think there are two pathways to adding FD support:

- Update the existing Frame trait doc-comments to specify that the DLC can be up to 15 and that message length can be up to 64 bytes, also adding other optional trait methods like `is_brs` for bitrate switching and the like.
- Or, creating an separate trait `FdFrame` which implements the above.

I prefer the first method since it could be done without breaking existing uses of the `Frame` trait.