rust-embedded/embedded-hal #581: HAL-Bus SPI Exclusive Device Unsatisfied Traits

tcaxle, 2024-03-18:
Good morning,

I am trying to understand if the unsatisfied traits error I am receiving in my implementation is due to an error in my implementation, the platform HAL package, or the underlying embedded HAL code.

I am trying to use the embedded_hal_bus ExclusiveDevice struct to wrap an SPI bus into an SPI device.

The SPI bus I have initialised using the platform HAL package (`tm4c129x_hal`):

```rust
let mut spi0 = hal::spi::Spi::spi0(
    p.SSI0,
    (
        porta
            .pa2
            .into_af_push_pull::<hal::gpio::AF15>(&mut porta.control),
        porta
            .pa5
            .into_af_push_pull::<hal::gpio::AF15>(&mut porta.control),
        porta
            .pa4
            .into_af_push_pull::<hal::gpio::AF15>(&mut porta.control),
    ),
    hal::spi::MODE_0,
    1_u32.mhz(),
    &clocks,
    &sc.power_control,
);
```

I'm then wrapping that up in the `ExclusiveDevice` object (`cs0` is just a normal GPIO pin in output mode).

```rust
let mut spi0_dev0 = ExclusiveDevice::new_no_delay(
    &spi0, cs0
);
```

However, this won't compile when I try to use the `write()` method of the `ExlusiveDevice` object. (`message` is a `Vec` of `byte`s).

```
error[E0599]: the method `write` exists for struct `ExclusiveDevice<&Spi<SSI0, (PA2<AlternateFunction<AF15, ...>>, ..., ...)>, ..., ...>`, but its trait bounds were not satisfied
  --> examples/tiva-c-connected-launchpad/examples/spi_hello.rs:67:27
   |
67 |         let _ = spi0_dev0.write(message);
   |                           ^^^^^ method cannot be called due to unsatisfied trait bounds
   |
  ::: /home/***/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embedded-hal-bus-0.1.0/src/spi/exclusive.rs:19:1
   |
19 | pub struct ExclusiveDevice<BUS, CS, D> {
   | --------------------------------------
   | |
   | doesn't satisfy `_: Default<_>`
   | doesn't satisfy `_: _embedded_hal_blocking_spi_Write<_>`
   |
   = note: the full type name has been written to '.../target/thumbv7em-none-eabihf/debug/examples/spi_hello-b9cc2dd24018ba0e.long-type-5094367532974101285.txt'
   = note: the following trait bounds were not satisfied:
           `ExclusiveDevice<&tm4c129x_hal::spi::Spi<tm4c129x_hal::tm4c129x::SSI0, (PA2<AlternateFunction<AF15, PushPull>>, PA5<AlternateFunction<AF15, PushPull>>, PA4<AlternateFunction<AF15, PushPull>>)>, PA6<Output<PushPull>>, NoDelay>: embedded_hal::blocking::spi::write::Default<_>`
           which is required by `ExclusiveDevice<&tm4c129x_hal::spi::Spi<tm4c129x_hal::tm4c129x::SSI0, (PA2<AlternateFunction<AF15, PushPull>>, PA5<AlternateFunction<AF15, PushPull>>, PA4<AlternateFunction<AF15, PushPull>>)>, PA6<Output<PushPull>>, NoDelay>: tm4c129x_hal::prelude::_embedded_hal_blocking_spi_Write<_>`
```

What I can't work out is what needs to be changed so that the trait bounds are satisfied. Can anyone shed some light?

Dirbaio, 2024-03-18:
`tm4c129x_hal` needs updating, it's implementing the embedded-hal 0.2 traits, while embedded-hal-bus uses embedded-hal 1.0

tcaxle, 2024-03-18:
Thanks for the quick response. I will look into the source and see about raising a PR to fix this.

Please can you point me in the direction of the specific changes that will need to be accounted for in the update? Is there a change-log for the embedded-hal-bus versions or should I just dig through commit messages?

tcaxle, 2024-03-19:
Still really struggling to get this HAL implementation correct. Can anyone point me at a HAL crate that implements `SpiBus` correctly so I have something to crib from?

Dirbaio, 2024-03-19:
Embassy HALs implement both the 0.2 and 1.0 traits, should be a good reference since you can compare: https://github.com/embassy-rs/embassy/blob/main/embassy-nrf/src/spim.rs#L549-L601

tcaxle, 2024-03-20:
That's fantastic, thank you!