rust-embedded/embedded-hal #584: SpiDevice implementations in embedded-hal-bus don't provide a way to use active-high chip select

gary600, 2024-03-28:
I am using a SPI display which requires active-high chip select, but all of the `SpiDevice` implementations don't mention the polarity of their CS pin anywhere. Looking at the implementation, these implementations are all hard-coded to use active-low CS, which won't work for this display. Adding a way of specifying the chip select polarity would make this usable for this case.

burrbull, 2024-03-28:
```rust
pub struct ExclusiveDevice<BUS, CS, D, const CSHIGH = false> {
```
???

Dominaezzz, 2024-05-15:
Couldn't you just create an `InvertedPin` struct that implements the `OutputPin` trait and does the inversion?

This seems like a non issue to me.

eldruin, 2024-05-24:
As a matter of fact, I wrote an [inverted-pin](https://crates.io/crates/inverted-pin) crate that does exactly that.
In my experience, active-low CS is the overwhelming majority so I think using an inverting wrapper should be enough and there is no pressing need to support this in a dedicated way.
Closing. Feel free to reopen if you think otherwise.

closed 2024-05-24 by eldruin