rust-embedded/embedded-hal #598: embedded-hal-bus no longer builds with target thumbv6m-none-eabi

Tschrock, 2024-04-26:
Added `embedded-hal-bus = "0.2.0"` to my project and it failed to build.
Using target `thumbv7m-none-eabi` works, using target `thumbv6m-none-eabi` fails. Version 0.1.0 works.

Trying to build the latest `master` branch:
```
./embedded-hal-bus> cargo build --target thumbv6m-none-eabi
   Compiling embedded-hal v1.0.0 (/home/cyber/Documents/VSCode/embedded-hal/embedded-hal)
   Compiling portable-atomic v1.6.0
   Compiling embedded-io-adapters v0.6.1 (/home/cyber/Documents/VSCode/embedded-hal/embedded-io-adapters)
   Compiling embedded-io-async v0.6.1 (/home/cyber/Documents/VSCode/embedded-hal/embedded-io-async)
   Compiling embedded-hal-async v1.0.0 (/home/cyber/Documents/VSCode/embedded-hal/embedded-hal-async)
   Compiling nb v1.1.0
   Compiling embedded-io v0.6.1 (/home/cyber/Documents/VSCode/embedded-hal/embedded-io)
   Compiling embedded-hal-bus v0.2.0 (/home/cyber/Documents/VSCode/embedded-hal/embedded-hal-bus)
   Compiling critical-section v1.1.2
   Compiling embedded-can v0.4.1 (/home/cyber/Documents/VSCode/embedded-hal/embedded-can)
   Compiling embedded-hal-nb v1.0.0 (/home/cyber/Documents/VSCode/embedded-hal/embedded-hal-nb)
error[E0599]: no method named `compare_exchange` found for struct `portable_atomic::AtomicBool` in the current scope
   --> embedded-hal-bus/src/i2c/atomic.rs:122:14
    |
120 | /         self.bus
121 | |             .busy
122 | |             .compare_exchange(
    | |             -^^^^^^^^^^^^^^^^ method not found in `AtomicBool`
    | |_____________|
    | 

error[E0599]: no method named `compare_exchange` found for struct `portable_atomic::AtomicBool` in the current scope
   --> embedded-hal-bus/src/spi/atomic.rs:127:14
    |
125 | /         self.bus
126 | |             .busy
127 | |             .compare_exchange(
    | |             -^^^^^^^^^^^^^^^^ method not found in `AtomicBool`
    | |_____________|
    | 

For more information about this error, try `rustc --explain E0599`.
error: could not compile `embedded-hal-bus` (lib) due to 2 previous errors
```

Seems to be caused by #593 

https://github.com/rust-embedded/embedded-hal/blob/a0ccb6579669804e9be723912ed645a16362c8a6/embedded-hal-bus/src/spi/atomic.rs#L125-L133

https://github.com/rust-embedded/embedded-hal/blob/a0ccb6579669804e9be723912ed645a16362c8a6/embedded-hal-bus/src/i2c/atomic.rs#L120-L128

ARMv6-M does not have atomic CAS, so `portable_atomic` does not implement `AtomicBool::compare_exchange` on this target.

Rahix, 2024-04-27:
I guess the most straight forward solution to fix this regression would be to add a feature-flag for `AtomicDevice`.  We did the same thing back in `shared-bus` to cater for platforms without the necessary atomics support.

Another solution would be to allow using `AtomicDevice` on single-core microcontrollers by use of critical sections instead of real atomics.  Not sure how best to expose this API-wise, though.

Rahix, 2024-04-27:
Also, maybe a lesson learned from this regression could be that we need to set up CI infrastructure in `embedded-hal` to check builds on all relevant targets?  There may be more subtle incompatibilities between platforms that could hit in this way, I think.

dan-corneanu, 2024-04-27:
I have stumbled upon this issue right now. Is there any workaround for the time being?

adamgreig, 2024-04-27:
Because AtomicDevice uses the portable-atomic crate, you can enable support for CAS on targets that don't normally have it with either the critical-section or unsafe-assume-single-core features: https://crates.io/crates/portable-atomic#optional-features-critical-section

Having to add an explicit dependency on portable-atomic and enable the feature just to get this crate to build isn't great, but it should work as a workaround now.

adamgreig, 2024-04-30:
We discussed this in today's meeting and the plan is:

* Add a new atomic-device feature, default off
* Only enable AtomicDevice when either the atomic-device feature is enabled or cfg(target_has_atomic...) is set, so it's by default available on platforms with CAS
* Add new portable-atomic-critical-section and portable-atomic-unsafe-assume-single-core features which proxy to the corresponding portable-atomic features (so users don't have to depend on portable-atomic directly) _and_ enable our atomic-device feature
* Document these new features
* Release as 0.3

closed 2024-07-28 by Dirbaio

keks, 2024-08-25:
I am still struggling to build this on `thumbv6m-none-eabi`. It looks like v0.3 hasn't been published to crates.io?

keks, 2024-08-25:
I am using
```
embedded-hal-bus = { git = "https://github.com/rust-embedded/embedded-hal", package = "embedded-hal-bus", revision = "2acbd77"}
```
for now :)

NZRosto, 2025-01-14:
Any update on this? v0.3 has not been released and embedded-hal-bus still does not build on `thumbv6m-none-eabi` targets. Any features/changes holding up the release?

jamesmunns, 2025-01-21:
CC https://github.com/rust-embedded/embedded-hal/pull/644

Dirbaio, 2025-01-21:
embedded-hal-bus v0.3.0 releaesd

PR #607 (2024-07-28): Add embedded-hal-bus atomic-device features
Add atomic-device and portable-atomic features as described in [this comment](https://github.com/rust-embedded/embedded-hal/issues/598#issuecomment-2086463791) 
The tasks from the comment are copied below for convenience
>
- Add a new atomic-device feature, default off
- Only enable AtomicDevice when either the atomic-device feature is enabled or cfg(target_has_atomic...) is set, so it's by default available on platforms with CAS
- Add new portable-atomic-critical-section and portable-atomic-unsafe-assume-single-core features which proxy to the corre [... cut]