rust-embedded/embedded-hal #599: Split embedded-can?

reneherrero, 2024-04-30:
Hi,

I noticed that the embedded-can crate is [always pulling the nb crate](https://github.com/rust-embedded/embedded-hal/blob/a0ccb6579669804e9be723912ed645a16362c8a6/embedded-can/Cargo.toml#L16) even if it's not being used.

One option would be to add features.  Another would be to refactor the code and follow the same pattern as the embedded-hal crates (better in my opinion).

Thoughts?

Dirbaio, 2024-04-30:
the `nb` crate is tiny, it compiles in milliseconds. If you don't use the nb stuff, code from `nb` doesnt end up at all on the final binary (the linker doesn't link unused code). 

IMO the split is not worth the effort.

reneherrero, 2024-04-30:
Then why create an embedded-hal-nb crate?  Async for CAN is also in the works #585 ...

reneherrero, 2024-05-06:
According to the stm32 bxcan project, the embedded-can crate is [deprecated](https://github.com/stm32-rs/bxcan/blob/3fc7a0e81975d4f25e61e0da81cd9e7a5e969e81/CHANGELOG.md?plain=1#L20).

Is this the case?  Understanding the state of affairs is "complicated"...  even more so as far as [CAN FD](#578) is concerned.

Any insight would be much appreciated,

reneherrero, 2025-12-17:
stale thread

closed 2025-12-17 by reneherrero