rust-embedded/embedded-hal #601: feature request: embedded_io_async::try_read

dvdsk, 2024-05-15:
Usecase
Catching up to a byte stream when lagging behind without waiting for new bytes to come in.  Such as getting the last 10 bytes received.

Example:
Library code using a uart buffered by the Hal implementation (like with embassy). A read call is done with a buffer of 100 bytes. It may return 100 bytes read. Is the last byte in the buffer the last that was received? 

A second read call could block forever so we can only guess. The only way to be certain is to ensure buffer provided to read is larger then what was provided to the bufferd uart.

Alternatives:
Could join on a near zero delay and a read. This should work but is less readable (the intent is less clear).

Dirbaio, 2024-05-15:
You can use `ReadReady` for this.

dvdsk, 2024-05-15:
Totally missed that, thanks for pointing it out!

closed 2024-05-15 by dvdsk